openrazer/openrazer #1821: dependency cannot be fulfilled || openrazer-driver-dkms

Cuvy95, 2022-05-28:
### Distribution

openrazer 3.3.0
Debian 11

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello Community, 
at first i want to say that this is my first post at Github, sorry for any mistakes.
im come from Germany and i use Parrot Os since 3 weeks. 
Now i will install openrazer but everytime when i download the .deb file from the official Homepage and i want to install it i become these Message : 

"Fehler Abhängikeit nicht erfüllbar: openrazer-driver-dkms (=3.3.0-0)"

I dont know what i can do to install it. 
I hope anyone can help me :)

z3ntu, 2022-05-28:
Please explain how you're trying to install it.

Cuvy95, 2022-05-28:
I downloaded it at this page [https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta](url)
and then i double click on the .deb file.

z3ntu, 2022-05-28:
Yeah, that won't work. Please use the way "Add repository and install manually" that adds the repository to your apt repo list.

Cuvy95, 2022-05-28:
i have tried it now and now i become these error (see in the picture
[image]
)

z3ntu, 2022-05-28:
Try removing the `openrazer-meta` package completely and then reinstall it.

Cuvy95, 2022-05-28:
i uninstalled it , but now comes this 
[image]
:

z3ntu, 2022-05-28:
Not sure. Dpkg is always cryptic in what it does and I don't use Debian so please ask for help in some Debian or Parrot support channel.

Cuvy95, 2022-05-28:
OK thanks for your help :)

Etaash-mathamsetty, 2022-06-13:
> i uninstalled it , but now comes this [image] :

its complaining that openrazer-meta is already not installed, so just manually remove openrazer-driver-dkms openrazer-daemon and python3-openrazer
`sudo apt remove openrazer-driver-dkms openrazer-daemon python3-openrazer`
then try (it should work if you added the repository)
`sudo apt install openrazer-meta`

closed 2022-11-11 by github-actions[bot]