openrazer/openrazer #1822: Razer Viper Ultimate erratic mouse sensitivity

rantnar, 2022-05-31:
I have this problem with my mouse. Both dock and mouse are recognized by openrazer and are working. But for some reason mouse sens ich constantly changing. I tried ubuntu, arch, fedora. x11 and wayland. Still same behaviour. Mouse dongle is connected to the dock.

lah7, 2022-06-03:
Have you tried without OpenRazer installed, a different system, or with Razer Synapse on Windows?

I don't really know what on OpenRazer's side could cause this, as it's only adjusting hardware parameters. Could it be a fault with the mouse, or maybe just a dirty sensor?

`libinput` could _potentially_ be a culprit on the software side, but only if this problem started happening recently with a system update.  #1690 found that an update started causing erratic scrolling.

On the hardware side, a higher polling rate means more CPU polling, so there is the possibility there is a bottleneck in your system or USB bus. Have you tried a different port or unplugged other USB devices?

rantnar, 2022-06-04:
It's fixed. I'm using UNRAID and passing through usb devices to a VM. It was my VM configuration that was problematic.. My usb controller was set to 3.0 qemu XHCI. For some reason it is problematic on linux VM, not on windows VM. Setting usb controller to 2.0 ehci fixes the issue with mose sens and polling rate. Sorry to bother you.

closed 2022-06-04 by rantnar

lah7, 2022-06-04:
Good to hear! Thanks for feeding back.

For what it's worth, if Razer Synapse ever offers a firmware upgrade, say **no**. @z3ntu ended up with a bricked Razer device because of USB passthrough to a VM.