openrazer/openrazer #1823: Support for Razer Blade 14'' with 3070Ti (2022)

turowicz, 2022-06-01:
### Device Info

* Name: Razer Blade 14'' with 3070Ti (2022)
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:028c Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028c 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 cze  1 13:20 .
drwxr-xr-x 4 root root 540 cze  1 13:20 ..
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if00 -> ../event4
lrwxrwxrwx 1 root root  10 cze  1 13:20 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if02 -> ../event17
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Razer_Razer_Blade-event-if01 -> ../event8
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Razer_Razer_Blade-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Razer_Razer_Blade-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 cze  1 13:20 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

turowicz, 2022-06-01:
Is the USB protocol sniffing required for a Laptop device? It's not like its sending anything anywhere over USB.

fbaumgardt, 2022-06-02:
Just adding a +1 for this model.

And @turowicz I replaced the Wifi card with an Intel AX210 to work around the problem you mention.

z3ntu, 2022-06-18:
@turowicz For RGB support you can probably just copy paste what previous Razer Blade additions did, assuming there's nothing new and weird going on with this laptop which I don't think is the case.

synacktic, 2022-07-06:
Hey all, I could create a pcap, but what actions do you want/need to see?

I have read over the info in the wiki, but want to know:
Should I just flip between the presets for lighting options? 
Do I need to go into chroma studio and light each key one at a time? 
How should I link what I'm doing to what is going on in the capture to make your lives easier?

z3ntu, 2022-07-23:
Re-opening as the device is not supported yet. @turowicz If you're no longer interested in this, you can just unsubscribe from the issue on the right side.

z3ntu, 2022-07-23:
@synacktic 

> Should I just flip between the presets for lighting options?

Basically this, probably 30 seconds of clicking around in the Synapse UI is enough.

synacktic, 2022-08-02:
Hey all, my Razer started having issues about 3 weeks after purchase, so I decided I did not trust the machine any returned it. So, I won't be able to capture the usb data.

pramodhrachuri, 2022-11-03:
Any estimate on when will the device be supported?

fbaumgardt, 2022-11-03:
I tried getting pcap but didn't get Synaptic in the VM to recognize the machine. Not sure what I missed, but I didn't have time to troubleshoot.

pramodhrachuri, 2022-12-19:
@sq1000000 found this in the wiki
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

alkeryn, 2022-12-19:
Hey, just wanted to tell you guys i got it running on nixos, i plan to have a windows vm anyway so i could make you a pcap in a few weeks!

alkeryn, 2022-12-20:
well for some reason razer synapse just gives a black screen in the windows vm so no idea what to do with it.

sq1000000, 2022-12-21:
@alkeryn I had the same problem when I tried Synapse 3 in Windows 11 in Virtualbox. The Synapse app works in qemu though.

sq1000000, 2022-12-21:
[image]
I got a setup running with wireshark and virt-manager, but I'm in way over my head here. I have no idea how to interpret the overwhelming amount of wireshark data. What do I do now?

Maybe, instead of capturing data and forging a new driver, is possible to just replicate the driver from the 2021 Blade 14 (1532:0270) and see if it works on this? They have the same form factor, and likely use a similar (if not the same) keyboard driver.

alkeryn, 2022-12-21:
@sq1000000 hey i also thought it was video related, so did you do a gpu passthrough for it to work ?

sq1000000, 2022-12-21:
@alkeryn I don't think it's video related. I'm just running a basic qemu virtual machine with [spice-guest-tools](https://www.spice-space.org/download.html) as the video driver. The keyboard is running through usb passthrough to the vm.

alkeryn, 2022-12-21:
@sq1000000 nice, well it is already better than nothing in the case you just want to set it to a static color.
also, maybe with windows iot version or whatnot you could run it in a sub 200MB VM so now we can start talking in the meanwhile.

though, if you can make a pcap by playing around that may be enough data to just update openrazer to support this new keyboard.

sq1000000, 2022-12-22:
@alkeryn Not 100% sure I did this right, but here's a pcap of me just clicking around in synapse. I changed the brightness of the keyboard, toggled the logo light, and changed the color of the keyboard a few times.

[razerblade14_synapse_1.zip](https://github.com/openrazer/openrazer/files/10282666/razerblade14_synapse_1.zip)

pramodhrachuri, 2022-12-29:
@sq1000000 Thanks for the pcap. Looks the data fragment matches with what is written in the [wiki](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol#phase-4---decoding-the-protocol).
I have messaged about this issue on the [community chat](https://github.com/openrazer/openrazer/wiki/Community-chat). Hopefully someone can add support soon.

I have another idea- I was looking through the [commit](https://github.com/openrazer/openrazer/commit/b91dbee1f66a29bc5bd8be3d21396eef25d1cf67) that added support for Razer Blade 14 (2021). Looks like add they had to do is to add the device IDs. Can someone writeup a patch for adding our device's ID and test it?

pramodhrachuri, 2022-12-29:
Also, how are you even plan on running this? Last time I tried, I faced some issue with installing due to the 6.0.1 kernel that I am running for using BT and WiFi. I don't remember the exact error prompt though. It was something about dkms or something.
How are you installing this?

sq1000000, 2022-12-29:
@pramodhrachuri I've installed it a few times on a few different distros. The installation is pretty straightforward. 

 You can see more here: https://openrazer.github.io/#download

pramodhrachuri, 2022-12-29:
Got it. I was able to compile and install it anyways.
Now, I was trying to make some changes as per the way I was mentioning and install it. And my python environment is now nuked!!!
I think this line is to blame -> https://github.com/openrazer/openrazer/blob/v3.5.1/Makefile#L170
Not sure how to restore it. Unfortunately apt is not working as well :/

### Python problem
I know this is tangential to this issue. But any resource to restore my python env will be highly appreciated.

I saw the following when I ran `sudo make ubuntu_install`.
[image]

I see the following when I try to run python3 now
```
➤ python3 
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.platlibdir = 'lib'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python310.zip',
    '/usr/lib/python3.10',
    '/usr/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f72b4ed0000 (most recent call first):
  <no Python frame>
```

z3ntu, 2022-12-29:
Well if you could tell me how I can discourage users from running these Makefile targets manually, you'd help me... 4 lines above that is a nice warning including a giant warning at the top of the Makefile...

Anyways first of all this manual moving will be gone with the next OpenRazer version since I found a better way of doing this. But for fixing your Python environment you should make sure there's nothing in `/usr/lib/python3.*` and all the nice packages are in `/usr/lib/python3/dist-packages` I believe. Debian/Ubuntu is very different there compared to every other distro for some reason. But not sure why your Python output doesn't list `/usr/lib/python3/` in `sys.path`. I guess you could also download some of the python3 deb files directly and do `dpkg -i foo.deb`

pramodhrachuri, 2022-12-29:
Thanks for the prompt reply!
I was able to fix it. I had to move the python3.10 folder outside the python3 folder.

And yes, it was my fault for directly using the target. Sorry about causing this panic due to my stupidity!

pramodhrachuri, 2022-12-29:
I have built the repo using the instructions in the [wiki](https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint). 
When I run the daemon, I see that "RazerBlade142022" is loaded but not getting detected. Maybe I missed a line or a file. Will try again tomorrow.
My patch file is here -> [f8ac05.txt](https://github.com/openrazer/openrazer/files/10318249/f8ac05.txt)

```
➤ openrazer-daemon -Fvr
No pidfile found, assuming openrazer-daemon is not running
Starting daemon.
2022-12-29 02:55:23 | razer                          | INFO     | Initialising Daemon (v3.5.1). Pid: 97721
2022-12-29 02:55:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2022-12-29 02:55:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
[... cut ...]
2022-12-29 02:55:23 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2022-12-29 02:55:23 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2022-12-29 02:55:23 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2022-12-29 02:55:23 | razer                          | INFO     | Serving DBus
2022-12-29 02:55:34 | razer                          | DEBUG    | DBus called get_serial_list
```

z3ntu, 2022-12-29:
Daemon changes are sufficient for now at least, the device is listed. I'm guessing now either driver changes are missing or they haven't been loaded yet. For keyboards/mice/etc you normally just unplug the USB cable and replug, you have to reboot I guess so the new modules get properly loaded.

Also I'm serious that if you have any suggestion on how to prevent people from running these Makefile targets, any help would be appreciated. And next time please follow this wiki page https://github.com/openrazer/openrazer/wiki/Building-a-package

pramodhrachuri, 2022-12-29:
I have rebooted my laptop and running `device_manager.devices` shows my keyboard. I will test it using Polychromatic and update my observations here,

Regarding the makefile-
1. I would suggest renaming the target names. In my case, I never even looked at the Makefile before running the command. I use fish-shell and hitting the tab after typing `make` or `sudo make`  lists the targets. I presumed `ubuntu_install` is the target for installing just by looking at the target name and never opened the Makefile. Also, I have a habit of just using the targets because almost every other software I recompiled from the source tells me to do that (for their repo).

2. Add a warning message that requires user confirmation before running something that might change their python environment. But this won't still stop people like me who are hastily trying out something. If you follow LTT on youtube you might know [what I am talking about](https://youtu.be/0506yDSgU7M?t=635) XD.

3. Add the building and installation instructions to the main README file in the home of the repo. Or at least put a link to the wiki. Right now, the installation section redirects me to the website which shows commands to install using apt. No mention of building from the source.

pramodhrachuri, 2022-12-29:
Polychromatic works on my laptop now. So, the [patch](https://github.com/openrazer/openrazer/files/10318249/f8ac05.txt) works. Just had to reboot my machine.

I can now control the colors of my keyboard. Yay!

alkeryn, 2022-12-31:
@pramodhrachuri hey that's great news ! :)
good work !

could you make a pull request ?

pramodhrachuri, 2022-12-31:
I made a [pull request](https://github.com/openrazer/openrazer/pull/1995) but the check for autopep8-formatting is failing.
The [logs](https://builds.sr.ht/~z3ntu/job/914271) are raising warnings on the py files I didn't even touch.

alkeryn, 2023-01-01:
@pramodhrachuri from what i understand you should use https://pypi.org/project/autopep8/ to format your code before pushing, not sure that's the issue but that's my guess at it.

pramodhrachuri, 2023-01-02:
@alkeryn yes. There is already a script in the repo that does it. `./scripts/format_source.sh`.
When I executed it, no changes were made.
In fact `./scripts/ci/check-autopep8-formatting.sh` also didn't report any error as shown in the [logs of auto checker](https://builds.sr.ht/~z3ntu/job/914380#:~:text=./scripts/ci/check%2Dautopep8%2Dformatting.sh).

pramodhrachuri, 2023-01-02:
Looking at the conversations on other PRs, I think someone (other than me) should test if the PR works and confirm it. Can anyone give it a try?

alkeryn, 2023-01-03:
@sq1000000 how did you managed to run synapse in the vm, i tried both a win10 and win11 qemu vm, none worked.
i just get a blank window.

do you have any idea ?
what display driver did you use ?
did you install drivers on windows ?

sq1000000, 2023-01-03:
@alkeryn I honestly have no Idea how I did it. I just installed the spice-guest-tools display driver (on the windows guest) and uninstalled it, and by some stroke of luck synapse started working.

If you need more packet captures, I can gladly supply them!

alkeryn, 2023-01-03:
@sq1000000 dang, could you give me your vm settings, like qemu command line or libvirt xml ? :)

sq1000000, 2023-01-03:
@alkeryn Sure. Here's a bunch of vm relevant stuff:

XML (rename txt to xml for it to work):
[win11.txt](https://github.com/openrazer/openrazer/files/10340339/win11.txt)

Device manager and Display driver info:
[Screenshot from 2023-01-03 18-09-19.png](https://user-images.githubusercontent.com/39637438/210456815-160f09ff-70fa-4ab6-8c7b-7f179d8c4c3d.png)

Spice guest tools exe:
https://www.spice-space.org/download/windows/spice-guest-tools/spice-guest-tools-latest.exe

pramodhrachuri, 2023-01-04:
An off-topic discussion.
Razer announced that our laptop model now supports USB4 (compatible with older versions of thunderbolt). However, both bare-metal win 11 and Ubuntu showed only USB 3 for me. So, when I contacted Razer customer support, they provided this PDF -> https://rzr.to/BQ9FhA with instructions to update to the new EC firmware v1.04. This firmware is not shown on the official page. IDK why.
While it is very tempting to install this, I haven't done it yet. I am scared of features breaking. Mainly WiFi and BT.

I thought to share it here as this thread has the highest number of users of this laptop I have seen anywhere.

alkeryn, 2023-01-04:
@pramodhrachuri i did make sure that the firmware was up to date before installing linux, you could see usb 4 in device manager, afaik windows update could do the firmware update, but that's a nice info, indeed it's kinda sad they didn't give it to everyone.

alkeryn, 2023-01-04:
@sq1000000 thanks ! :)

fbaumgardt, 2023-01-04:
I installed your repo and it works. I'm using Polychromatic and so far all the settings work as excited. Awesome work, thank you!

pramodhrachuri, 2023-01-05:
@alkeryn  I think this happened because you have a newer batch of the laptop. Unfortunately windows update didn't help with USB 4 in my case. Idk why.

The patched WCN firmware working with EC firmware is all I wanted to hear. Thank you so much!

closed 2023-01-15 by z3ntu