openrazer/openrazer #1825: OPENRAZER: ERROR

PlamenStilyianov, 2022-06-02:
Operating System: openSUSE Tumbleweed 20220531
KDE Plasma Version: 5.24.5
KDE Frameworks Version: 5.94.0
Qt Version: 5.15.2
Kernel Version: 5.17.9-1-default (64-bit)
Graphics Platform: X11
Processors: 20 × 12th Gen Intel® Core™ i9-12900H
Memory: 62.5 GiB of RAM
Graphics Processor: Mesa Intel® Graphics

ERROR:

openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Thu 2022-06-02 21:10:10 BST; 24s ago
       Docs: man:openrazer-daemon(8)
    Process: 3631 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 3631 (code=exited, status=1/FAILURE)
        CPU: 71ms

z3ntu, 2022-06-02:
Please run `openrazer-daemon -Fv` in the terminal and post the output.

PlamenStilyianov, 2022-06-02:
orajava@suse154:~> openrazer-daemon -Fv
Starting daemon.
2022-06-02 21:20:51 | razer                          | CRITICAL | User is not a member of the plugdev group
2022-06-02 21:20:51 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Stopping daemon.

z3ntu, 2022-06-03:
Please read the output and do what's written there.

closed 2022-11-02 by github-actions[bot]