openrazer/openrazer #1826: Update instructions for Fedora

paidhi, 2022-06-04:
Fedora 33 is EOL and Fedora 34 is going [EOL in a few days](https://ask.fedoraproject.org/t/fedora-34-is-going-eol-in-one-week/22924) (2022-06-07). Also Fedora 36 was released.
Could you please update https://openrazer.github.io/#download (remove Fedora 33/34 and add Fedora 36)?

paidhi, 2022-06-04:
I can confirm that the packages for `openrazer-meta` and `polychromatic` work fine with Fedora 36 ("Razer Huntsman Tournament Edition keyboard").
- `sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_36/hardware:razer.repo`
- `sudo dnf install openrazer-meta`

lah7, 2022-06-04:
Thank you! Instructions updated.

closed 2022-06-04 by lah7