openrazer/openrazer #1827: Ripple effect not working

nab138, 2022-06-04:
### Distribution
Linux mint 20.3 Una (Ubuntu based)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The ripple effect no longer works. Every other effect works (including reactive), but if I choose the ripple effect in razerCommander or polychromatic it doesn't do anything when I press a key. It used to work fine, and I have no idea what happened. It's been like this for a few months now. I've tried to reinstall openrazer-daemon and openrazer-driver-dkms. Tried to manually run `python3 -c "import openrazer.client;a=openrazer.client.DeviceManager();a.devices[0].fx.ripple_random()"` but I get `No module named 'openrazer.client'
`

nab138, 2022-06-04:
I dual boot the same machine with windows and the ripple effect works fine there, so It can't be a hardware issue of any kind.

lah7, 2022-06-04:
Ripple effect is technically a software effect powered by the daemon process. Are you able to create custom lighting in Polychromatic ("Effects" tab) or razerCommander ("Custom" effect)?

**Also:**

* Which keyboard do you have?
* Which version of OpenRazer? Just want to check as there is a version in Ubuntu's repositories that might be out of date:

      openrazer-daemon --version

nab138, 2022-06-04:
Daemon version: 2.5.0
Keyboard: Razer Huntsman Elite
I am able to create custom effects in razerCommander. I've uninstalled polychromatic in favor of razercommander, so I can't try it there.

lah7, 2022-06-04:
2.5.0 is the one packaged by Debian. Please install the latest (3.3.0) packaged by us, and reboot the system:

    sudo add-apt-repository ppa:openrazer/stable
    sudo apt update
    sudo apt install openrazer-meta

-- https://openrazer.github.io/#ubuntu

nab138, 2022-06-04:
Will try now, thank you

nab138, 2022-06-04:
`openrazer-daemon --version` now outputs 3.3.0, but the ripple effect still doesn't work (yes I rebooted)

nab138, 2022-06-04:
I suspect it has something to do with python, considering it is a software effect. Python is kind of a mess on my computer, I had versions install through linuxbrew that messed with my system and I think I resolved it but I'm not quite sure

lah7, 2022-06-04:
Okay, that rules out the old version being an issue. The only bug I know is that changing the brightness while the ripple is running will cause it to freeze (#1679).

Since you confirmed creating a custom effect works, there doesn't seem to be anything wrong with the protocol. It can only mean the daemon isn't detecting key presses or the ripple thread crashes for some reason, assuming the keyboard is left in darkness when it's activated.

Please clear the log and restart the daemon:

    rm ~/.local/share/openrazer/logs/razer.log
    openrazer-daemon -rFv

Then try activating the ripple effect and press some keys. Please attach the log file `~/.local/share/openrazer/logs/razer.log` here in case it is crashing.

You could try replugging the keyboard (and activate ripple again), just in case something is initializing in the wrong order at boot (I believe the keyboard is exposed as 2 input devices, as seen `xinput list`)

lah7, 2022-06-04:
Whoops, didn't see the reply just minutes ago. ^^ Yes, it will be software, and if your Python installation is a mess, that could be why.

One suggestion is to check the PATHs all point to system libraries:

    echo $PATH
    python3 -c "import sys; print(sys.path)"

Then compare with a clean Linux Mint installation (or live session). That said, if this `linuxbrew` installed into the system libraries folders, then there's probably a utility out there that can delete files that don't belong to any package in  `/usr/lib/python3/` (etc).

nab138, 2022-06-04:
The only thing of interest I saw in the openrazer daemon logs was:
`2022-06-04 11:47:04 | razer.device0                  | DEBUG    | DBus call set_brightness
2022-06-04 11:47:04 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerHuntsmanElite:PM2029F07306397, 'setBrightness', 255)
2022-06-04 11:47:04 | razer.device0                  | DEBUG    | DBus call set_ripple_effect
2022-06-04 11:47:04 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerHuntsmanElite:PM2029F07306397, 'setRipple', None, None, None, dbus.Double(0.05))
`
The whole log is found at https://hatebin.com/adazjphfnr

The keyboard is in fact, left in the dark when enabling the ripple effect.

I would prefer not to replug the keyboard, as I have an all in one machine, the port is in the back behind a cover I can't access without pulling out most other cables there.

Output of those path commands:

`                        /home/nicholas/.fly/bin:/home/nicholas/go/bin:/home/nicholas/.local/bin:/home/nicholas/CEdev/bin:/home/nicholas/.nvm/versions/node/v16.13.1/bin:/home/linuxbrew/.linuxbrew/bin:/home/linuxbrew/.linuxbrew/sbin:/home/nicholas/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-16-oracle/bin:/usr/lib/jvm/java-16-oracle/db/bin:/home/nicholas/coding/fabricmods/GeneratorFabricMod/bin:/usr/local/go/bin`

`['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/nicholas/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.8/dist-packages']`

I will try a live usb in a bit assuming these outputs don't help find a solution

nab138, 2022-06-04:
From linux mint on distrotest.net (More convienient then booting the live usb)

echo $PATH:
`/usr/local/sbin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin`

The python command is identical except it is lacking the `'/home/nicholas/.local/lib/python3.8/site-packages'`

lah7, 2022-06-05:
[Python virtual environments](https://docs.python.org/3/tutorial/venv.html) are recommended if you intend to develop/install Python applications not provided via Debian packages. Installing stuff in `/usr/local` and `~/.local` ends up interfering (+ overriding) system packages and libraries.
 
Can't say for certain the mess is the cause, It seems likely considering your initial comment says "_it use to work fine"_ and "_been like this for a few months_". One way you **might** be able to test is to create a new user and run `openrazer-daemon` under a fresh profile. This can't 100% confirm since there could be leftover in `/usr/local` and other paths.

Are you using `linuxbrew`? If not, I would recommend inspecting what's installed in your home directory, and other custom paths, deleting them if necessary:

* `~/.local/bin`
* `~/.local/lib/python3.8/site-packages`
* `/home/linuxbrew`
* `/usr/local/bin`

Another test could be to install/run Linux Mint on another USB drive or partition and install OpenRazer. If it works on the same hardware in a fresh environment, we'll know for sure.

closed 2022-11-04 by github-actions[bot]

campbellcole, 2023-02-11:
I am currently having this issue as well. Every other effect works. I have a Huntsman TE.

Something interesting I noticed that maybe indicates something, when using the Device Info > Inspect Matrix with Ripple selected, clicking a grid cell only flashes the key white for a few milliseconds then turns it back off, while with any other effect, the key stays lit.

nab138, 2023-02-19:
Out of interest, did/do you also have linuxbrew installed (Or is your python installation messy)? If so that is probably the cause

campbellcole, 2023-02-20:
No I use NixOS and the installation was managed by the derivations. The Python environment openrazer has access to is essentially a venv with only the necessary packages installed. It is about as clean as it gets.

nab138, 2023-02-22:
I haven't gotten it to work, unfortunately. I'll let you know if I get any leads

nab138, 2023-06-18:
I've actually found what seems to be the problem because of #1974 , if I run ls in `/dev/input/by-id/` I get
```
usb-FIFINE_Microphones_FIFINE_K678_Microphone_REV1.0-event-if03
usb-Generic_HP_2.0MP_High_Definition_Webcam_200901010001-event-if00
usb-Logitech_USB_Receiver-if02-event-mouse
usb-Logitech_USB_Receiver-if02-mouse
usb-Melfas_LGD_AIT_Touch_Controller-event-if00
usb-Razer_Razer_Huntsman_Elite-event-kbd
usb-Razer_Razer_Huntsman_Elite-if01-event-kbd
usb-Razer_Razer_Huntsman_Elite-if02-event-mouse
usb-Razer_Razer_Huntsman_Elite-if02-mouse
```
and if I try to cat them all, NONE of the Razer devices output anything when I type (other devices do). I did however find that /dev/input/event24 does have output when I type. My razer keyboard types perfectly fine on this machine so idk what to do. Is there a way I could force it to use /dev/input/event24 instead of trying to use the razer huntsman symlink? or should I try to fix the symlink?

nab138, 2023-06-18:
I finally solved this! It was a software called input-remapper that I used to re-assign some of my keys, but it was interfering with the files in /dev/input/by-id which prevented the ripple effect from working.