openrazer/openrazer #1829: I have a supported device but razergenie says I don't.

DataBase137, 2022-06-05:


lah7, 2022-06-05:
It could mean an old version is installed, or OpenRazer wasn't installed properly. [(See the Troubleshooting guide if it's the latter)](https://github.com/openrazer/openrazer/wiki/Troubleshooting).

What version of OpenRazer? The latest right now is 3.3.0.

    openrazer-daemon --version

Which distro? Did you install it using the instructions on https://openrazer.github.io?

DataBase137, 2022-06-05:
I have 3.3.0 installed, and I'm on Fedora 36 Workstation Edition.

DataBase137, 2022-06-05:
So I actually installed it via the GNOME store, but I uninstalled it now, and installed it the way you have it on the guide, but now I don't have the application.

lah7, 2022-06-05:
Maybe you need to install RazerGenie again?

I noticed on the [Flatpak page](https://flathub.org/apps/details/xyz.z3ntu.razergenie), there is this:

> Requires OpenRazer to be installed outside of flatpak via a distribution package.

In theory, we should be on the right track. Unless there's [something else that's needs addressing](https://github.com/openrazer/openrazer/wiki/Troubleshooting) (like plugdev, secure boot, etc)

DataBase137, 2022-06-05:
Well if I install RazerGenie it says I don't have a supported device
[image]
.

lah7, 2022-06-05:
That must mean that OpenRazer isn't set up on your computer properly. Please try 2, 3 and 4 from this troubleshooting guide and let us know if you get stuck:

https://github.com/openrazer/openrazer/wiki/Troubleshooting

DataBase137, 2022-06-05:
With the 2nd option, I got this error:
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-3.3.0 does not exist.

DataBase137, 2022-06-05:
With the 3rd option, I got this error:
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.17.11-300.fc36.x86_64

lah7, 2022-06-06:
The driver isn't installed. Which distro are you using?

DataBase137, 2022-06-07:
Fedora 36 with GNOME 42.2

z3ntu, 2022-06-07:
Did you follow the installation instructions? https://openrazer.github.io/#fedora

DataBase137, 2022-06-07:
Yes, I did.

z3ntu, 2022-06-07:
`ls -al /usr/src/` ?

vola91, 2022-06-08:
`:~$ ls -al /usr/src/
总用量 28
drwxr-xr-x  7 root root 4096 6月   8 12:25 .
drwxr-xr-x 14 root root 4096 10月 23  2021 ..
drwxr-xr-x  7 root root 4096 10月 23  2021 linux-headers-5.11.0-38-generic
drwxr-xr-x  7 root root 4096 6月   6 16:27 linux-headers-5.13.0-44-generic
drwxr-xr-x 24 root root 4096 10月 23  2021 linux-hwe-5.11-headers-5.11.0-38
drwxr-xr-x 24 root root 4096 6月   6 16:27 linux-hwe-5.13-headers-5.13.0-44
drwxr-xr-x  3 root root 4096 6月   8 12:46 openrazer-driver-3.3.0`

Same device and same issue here with Polychromatic.    PID: 1532:009c

DataBase137, 2022-06-09:
I'm fine without razer for now.

closed 2022-06-09 by DataBase137

nightsky30, 2022-06-10:
> :~$ ls -al /usr/src/ 总用量 28 drwxr-xr-x 7 root root 4096 6月 8 12:25 . drwxr-xr-x 14 root root 4096 10月 23 2021 .. drwxr-xr-x 7 root root 4096 10月 23 2021 linux-headers-5.11.0-38-generic drwxr-xr-x 7 root root 4096 6月 6 16:27 linux-headers-5.13.0-44-generic drwxr-xr-x 24 root root 4096 10月 23 2021 linux-hwe-5.11-headers-5.11.0-38 drwxr-xr-x 24 root root 4096 6月 6 16:27 linux-hwe-5.13-headers-5.13.0-44 drwxr-xr-x 3 root root 4096 6月 8 12:46 openrazer-driver-3.3.0
> 
> Same device and same issue here with Polychromatic. PID: 1532:009c Ubuntu 20.04/Zorin 16 Tried on both statble and daily PPA.
> 
> Not supported yet? #1339 #1307

Both of those issues refer to the Basilisk X Hyperspeed and don't match the PID you provided. PID: 1532:009c seems to be the DeathAdder V2 X HyperSpeed. You may want to take a look at #1711

vola91, 2022-06-10:
> Both of those issues refer to the Basilisk X Hyperspeed and don't match the PID you provided. PID: 1532:009c seems to be the DeathAdder V2 X HyperSpeed. You may want to take a look at #1711 #1773

oh sorry for mixing that up. I have read these two issues but I am not sure if the driver of V2 X HyperSpeed is contained in V3.3.0. Because it said "The following PIDs are not supported in the lastest stable release..". I installed with Ubuntu 20.04 Stable PPA, then I switched to Daily PPA. Still can't make it works, I will leave a commit there, thank you

Bus 003 Device 015: ID 1532:009c Razer USA, Ltd Razer DeathAdder V2 X HyperSpeed