openrazer/openrazer #1833: Does this package fix webcam for the Razer Blade [1532:0255] ?

CharlieBrownCharacter, 2022-06-08:
### Distribution

Bus 001 Device 005: ID 1532:0255 Razer USA, Ltd Razer Blade

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Not really an issue but I would like to know if this package fixes the issue where the webcam on Razer Blade [1532:0255] doesn't work. I have found in several online places that Ubuntu doesn't support the webcam for Razer Blade.

For example, when I open Chesse I get this from `dmesg`

```
[   34.173164] input: USB Camera: USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input43
[   38.830501] kauditd_printk_skb: 12 callbacks suppressed
[   38.830505] audit: type=1326 audit(1654674873.585:77): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.slack.slack pid=3295 comm="slack" exe="/snap/slack/62/usr/lib/slack/slack" sig=0 arch=c000003e syscall=92 compat=0 ip=0x7f165c6763b7 code=0x50000
[  106.420662] usb 1-7: USB disconnect, device number 7
[  106.840175] usb 1-7: new high-speed USB device number 8 using xhci_hcd
[  107.017130] usb 1-7: New USB device found, idVendor=13d3, idProduct=56bd, bcdDevice=20.01
[  107.017142] usb 1-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  107.017147] usb 1-7: Product: USB Camera
[  107.017151] usb 1-7: Manufacturer: Azurewave
[  107.017155] usb 1-7: SerialNumber: 0x0001
[  107.021764] usb 1-7: Found UVC 1.00 device USB Camera (13d3:56bd)
[  107.025607] usb 1-7: Failed to initialize entity for entity 6
[  107.025616] usb 1-7: Failed to register entities (-22).
[  107.025832] input: USB Camera: USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input44
[  110.620831] usb 1-7: USB disconnect, device number 8
[  111.004204] usb 1-7: new high-speed USB device number 9 using xhci_hcd
[  111.184184] usb 1-7: New USB device found, idVendor=13d3, idProduct=56bd, bcdDevice=20.01
[  111.184187] usb 1-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  111.184188] usb 1-7: Product: USB Camera
[  111.184189] usb 1-7: Manufacturer: Azurewave
[  111.184189] usb 1-7: SerialNumber: 0x0001
[  111.187891] usb 1-7: Found UVC 1.00 device USB Camera (13d3:56bd)
[  111.191202] usb 1-7: Failed to initialize entity for entity 6
[  111.191204] usb 1-7: Failed to register entities (-22).
[  111.191238] input: USB Camera: USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input45
[  566.626080] usb 1-7: USB disconnect, device number 9
[  567.026247] usb 1-7: new high-speed USB device number 10 using xhci_hcd
[  567.203327] usb 1-7: New USB device found, idVendor=13d3, idProduct=56bd, bcdDevice=20.01
[  567.203340] usb 1-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  567.203345] usb 1-7: Product: USB Camera
[  567.203349] usb 1-7: Manufacturer: Azurewave
[  567.203352] usb 1-7: SerialNumber: 0x0001
[  567.207718] usb 1-7: Found UVC 1.00 device USB Camera (13d3:56bd)
[  567.211514] usb 1-7: Failed to initialize entity for entity 6
[  567.211522] usb 1-7: Failed to register entities (-22).
[  567.211717] input: USB Camera: USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input46
[  567.303017] usb 1-7: USB disconnect, device number 10
[  567.694256] usb 1-7: new high-speed USB device number 11 using xhci_hcd
[  567.875004] usb 1-7: New USB device found, idVendor=13d3, idProduct=56bd, bcdDevice=20.01
[  567.875017] usb 1-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  567.875022] usb 1-7: Product: USB Camera
[  567.875026] usb 1-7: Manufacturer: Azurewave
[  567.875030] usb 1-7: SerialNumber: 0x0001
[  567.879550] usb 1-7: Found UVC 1.00 device USB Camera (13d3:56bd)
[  567.883221] usb 1-7: Failed to initialize entity for entity 6
[  567.883230] usb 1-7: Failed to register entities (-22).
[  567.883497] input: USB Camera: USB Camera as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input47

```

I have tried for about 2 months to make this webcam work but I'm about to give up and go out and buy a new one.

Etaash-mathamsetty, 2022-06-08:
> ### Distribution
> Bus 001 Device 005: ID 1532:0255 Razer USA, Ltd Razer Blade
>
> ### Support
> * [x]  Have you tried rebooting?
> * [x]  Have you looked at the [...]

have you tried it on windows? If it doesn't work there, then go ahead and do that

CharlieBrownCharacter, 2022-06-08:
@Etaash-mathamsetty Thank you for the reply. I really appreciate it.

I have tried on Windows and it works there just perfect. This seems to be just an Ubuntu issue

z3ntu, 2022-06-08:
> Does this package fix webcam for the Razer Blade [1532:0255] ?

No. OpenRazer only handles lighting and some other functionality like that

Etaash-mathamsetty, 2022-06-08:
> @Etaash-mathamsetty Thank you for the reply. I really appreciate it.
> 
> I have tried on Windows and it works there just perfect. This seems to be just an Ubuntu issue

try using a different distro, maybe ubuntu's kernel is too old. (like usual lol)

CharlieBrownCharacter, 2022-06-12:
Bought a C270 HD webcam from Logitech. Working like a charm now.

Thank you for taking the time @z3ntu and @Etaash-mathamsetty

closed 2022-06-12 by CharlieBrownCharacter

Etaash-mathamsetty, 2022-06-12:
> Bought a C270 HD webcam from Logitech. Working like a charm now.
> 
> Thank you for taking the time @z3ntu and @Etaash-mathamsetty

did you at least try to use a custom kernel?

CharlieBrownCharacter, 2022-06-14:
@Etaash-mathamsetty yeah, tried but no luck there.

I have "lost" more than 2 months on back and forward trying to make the webcam work on my laptop but no luck. Better to spend 20 bucks on a webcam than to spend one more week on this issue.

I truly appreciate your suggestion though

Etaash-mathamsetty, 2022-06-14:
> @Etaash-mathamsetty yeah, tried but no luck there.
> 
> I have "lost" more than 2 months on back and forward trying to make the webcam work on my laptop but no luck. Better to spend 20 bucks on a webcam than to spend one more week on this issue.
> 
> I truly appreciate your suggestion though

have you looked here? (look under the webcam section)
https://wiki.archlinux.org/title/Razer_Blade