openrazer/openrazer #1836: Razer Atheris Excessive Battery Notifications

ixalsera, 2022-06-13:
### Distribution

Ubuntu 22.04 Jammy Jellyfish (GNOME 42.1 on Wayland)

### Support

- [ X ] Have you tried rebooting?
- [ X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

### Issue
Every so often, I receive an AppIndicator notification that my Atheris' battery level is at 0% despite the fact that it clearly isn't. At some stage it did report the battery at 30-ish percent - I presume correctly - but I haven't seen a non-0% notification for a while (Polychromatic says my battery is at 52% which matches the output of the debug.log: [https://pastebin.com/xuqQaVMm](https://pastebin.com/xuqQaVMm)).

Couldn't find any mention of this so thought I'd open a ticket. The debug.log pasted above is just the excerpt from today, I can paste a longer one if needed and/or any other outputs. `journalctl` didn't reveal anything.

Other notable things: Atheris is running off 2.4GHz dongle

lah7, 2022-06-13:
Hi. Is the issue about wanting to reduce the notification frequency, or that it reports 0%, or both?

For the former, take a look at `~/.config/openrazer/razer.conf` in a text editor and tweak the settings - the keys in red are old and should be deleted:

```diff
-mouse_battery_notifier = False
-mouse_battery_notifier_freq = 0
battery_notifier = True
battery_notifier_freq = 300
battery_notifier_percent = 33
```

* `battery_notifier_freq` is in seconds. 300 = 5 minutes.
* `battery_notifier_percent` is the threshold before notifications start to appear. Though if it reports 0%, then it's below the default `33`% and will always appear. 

---

For the latter, it is unknown why the daemon sometimes reads the battery % as zero: #1798

#1749 found the wrong % happens after a reboot. Does restarting the daemon (after closing frontends like Polychromatic) help "fix" the incorrect percentage until you next logon?

    systemctl --user restart openrazer-daemon.service

ixalsera, 2022-06-14:
> Hi. Is the issue about wanting to reduce the notification frequency, or that it reports 0%, or both?

A bit of both. Because it's reading the percentage as 0% then the notifications can be a little excessive.

> [T]ake a look at `~/.config/openrazer/razer.conf` in a text editor and tweak the settings

This actually explained a lot, thanks. Particularly, why I saw notifications below 33% and not currently at 53%. My config has `battery_notifier_freq` set to 600s which I think also correlates with how often I get notifications. So mark the frequency part as solved. :)

More importantly, I noticed the `devices_off_on_screensaver` setting which is set to true. Taking a look back over that debug log, I found these quite interesting:
```
2022-06-13 10:38:47 | razer.screensaver              | DEBUG    | Received screensaver active signal
2022-06-13 10:38:47 | razer.device0                  | INFO     | Suspending RazerAtherisReceiver
2022-06-13 10:38:47 | razer.device0                  | DEBUG    | Device doesn't have suspend/resume
2022-06-13 10:40:32 | razer.device0                  | DEBUG    | DBus call get_battery
2022-06-13 10:40:32 | razer.device0.batterynotifier  | DEBUG    | Razer Atheris (Receiver) Battery at 0%
2022-06-13 10:42:02 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2022-06-13 10:42:02 | razer.device0                  | INFO     | Resuming RazerAtherisReceiver
2022-06-13 10:42:02 | razer.device0                  | DEBUG    | Device doesn't have suspend/resume
```
Based on that, It looks like the issue might be when my lock screen starts and I've just never made the link before. The dongle is connected to an unpowered, 4-port mini hub and so I'm making an edumacated guess here that while the lock screen is active, Ubuntu may be powering down the hub and so battery reporting through the dongle is incorrect? 

I'm at home for the next few days and the Atheris is in the office so I can't do any checks right now, but I'll go a day or so with the dongle directly connected to the machine when I can and paste a new debug log then to see if there's any difference. I'll also set `devices_off_on_screensaver` to False and see if that makes any difference (doubtful since the device has no Suspend/Resume, but worth a test anyway I guess).

closed 2022-11-13 by github-actions[bot]