openrazer/openrazer #1839: Request: Support for Huntsman v2 1523:026B

0pLuS0, 2022-06-18:
### Distribution

Slackware 15.0

### Support

- [X ] Have you tried rebooting?
- [X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I believe that Huntsman v2 1523:026B is not supported, I didn't see it listed, and wanted to ask, will there be support for it soon? I just received the keyboard today, so I really hope there will be support soon.

Here's a screen shot in Polychromatic showing it failed.

[image]

```THANKS```

z3ntu, 2022-06-18:
Duplicate of #1681

closed 2022-06-18 by z3ntu

z3ntu, 2022-06-21:
The device is not yet supported as indicated in the linked issue. However there's an open pull request for this device, you can try building that and testing it. Instructions you can find in the wiki "building a package" the page is called

0pLuS0, 2022-06-29:
Ok, any chance of knowing when it's expected?

THANKS

z3ntu, 2022-07-01:
Whenever it's ready.