openrazer/openrazer #1841: Support for Razer Viper V2 Pro mouse

Karagra, 2022-06-19:
### Device Info

* Name: Razer Viper V2 Pro
* Product Number: R201-04390100
* Model Number: R201-0439

### Device Information
```
Bus 003 Device 002: ID 1532:00a6 Razer USA, Ltd Razer Viper V2 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a6 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Viper V2 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```

```
Please post the output of
total 0
drwxr-xr-x 2 root root 360 Jun 19 10:54 .
drwxr-xr-x 4 root root 740 Jun 19 10:54 ..
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-ACTIONS_Pebble_V3-event-if03 -> ../event9
lrwxrwxrwx 1 root root  10 Jun 19 10:54 usb-ASRock_LED_Controller_A02019100900-event-joystick -> ../event28
lrwxrwxrwx 1 root root   6 Jun 19 10:54 usb-ASRock_LED_Controller_A02019100900-joystick -> ../js0
lrwxrwxrwx 1 root root  10 Jun 19 10:54 usb-Razer_Razer_Huntsman_V2_Analog-event-if04 -> ../event26
lrwxrwxrwx 1 root root  10 Jun 19 10:54 usb-Razer_Razer_Huntsman_V2_Analog-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Jun 19 10:54 usb-Razer_Razer_Huntsman_V2_Analog-if01-event-kbd -> ../event24
lrwxrwxrwx 1 root root  10 Jun 19 10:54 usb-Razer_Razer_Huntsman_V2_Analog-if02-event-mouse -> ../event25
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Huntsman_V2_Analog-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-event-if03 -> ../event8
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-if01-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Jun 19 10:54 usb-Razer_Razer_Viper_V2_Pro_000000000000-mouse -> ../mouse0

```

### Packet Captures

Not sure if this is needed and looks complicated but will do the work if required.

### Synapse Screenshots
https://imgur.com/a/HcDtC9X

Karagra, 2022-06-19:
I am confused you said for me to make a PR but the preset for Add New Device is under issues. Is this the correct place top put this?

Etaash-mathamsetty, 2022-07-12:
> I am confused you said for me to make a PR but the preset for Add New Device is under issues. Is this the correct place top put this?

if you know how to code and contribute to the project to add support for this device, then you can make a PR, otherwise leaving this issue as-is is fine.

z3ntu, 2022-07-23:
Duplicate of #1820 

Thanks @Etaash-mathamsetty

PR #1849 (2022-07-23): driver: add support for Razer Viper V2 Pro
Tested both wired and wireless with polychromatic and looks like everything is working.

Some wireshark captures for this device https://gist.github.com/blacha/d4bee6ea474a86258a39028c6ba1679d

Fixes #1841 Fixes #1820
closed 2022-07-23 by z3ntu