openrazer/openrazer #1843: Having problems getting headers to use with openrazer

CobraA1, 2022-06-23:
### Distribution

Manjaro 21.0

### Support

- [ ] Have you tried rebooting?

Yes

- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Yes

The troubleshooting page indicates I am missing linux-headers, but I am unable to obtain them. Would like advice on how to approach this and who to contact.

[image]

z3ntu, 2022-06-24:
Update your system (pacman -Syu) and then it should work

CobraA1, 2022-06-24:
> Update your system (pacman -Syu) and then it should work

That fixes it, thanks!

Perhaps make a note on the [troubleshooting page](https://github.com/openrazer/openrazer/wiki/Troubleshooting)?

z3ntu, 2022-07-07:
Tbh that's not a common problem and definitely not at all related to openrazer. Any time you're getting a 404 error with pacman either your mirror is quite broken or you haven't updated your package lists. And partial updates have also never been supported in Arch Linux and I assume neither in Manjaro.

closed 2022-07-07 by z3ntu