openrazer/openrazer #1854: DeathAdder V2 x Hyperspeed - Unrecognised

jrwagh333, 2022-07-17:
OpenRazer detects a device but has no idea what it is. I see that the BlackWidow Hyperspeed and other Hyperspeed devices are supported, but unfortunately not the DeathAdder. 

Please could somebody provide directions on how I can provide the necessary information to get the device added to the software in an update.

z3ntu, 2022-07-17:
Please provide output of `lsusb | grep 1532`

Etaash-mathamsetty, 2022-07-18:
> OpenRazer detects a device but has no idea what it is. I see that the BlackWidow Hyperspeed and other Hyperspeed devices are supported, but unfortunately not the DeathAdder.
>
> Please could somebody [...]

I am pretty sure this mouse is already supported, it's just that open razer hasn't got a new release with support for the mouse

jrwagh333, 2022-07-18:
> I am pretty sure this mouse is already supported, it's just that open razer hasn't got a new release with support for the mouse

I can't see it listed on the website

Etaash-mathamsetty, 2022-07-18:
> I can't see it listed on the website

check readme.md under supported mice

jrwagh333, 2022-07-21:
Hmm, thanks for that. Polychromatic sees it as an Unsupported Device though.

z3ntu, 2022-07-23:
@jrwagh333 It's not part of the latest release v3.3.0, only supported in git master yet. On Ubuntu you can install the daily packages, or on Arch Linux openrazer-git. For other distros you can build the package from source youself: https://github.com/openrazer/openrazer/wiki/Building-a-package

closed 2022-07-23 by z3ntu