openrazer/openrazer #1858: Static effect on Lambda Labs x Razer Tensorbook (Razer Blade 15 Advanced Mid 2021) does not respect color

kkraus14, 2022-07-24:
### Distribution

PopOS 22.04 LTS

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm on a Lambda Labs x Razer Tensorbook (https://lambdalabs.com/deep-learning/laptops/tensorbook). The USB ID matches the Razer Blade Advanced Mid 2021:
```
Bus 003 Device 003: ID 1532:0276 Razer USA, Ltd Razer Blade
```
That being said, the default backlight color of the keyboard is pink as opposed to the normal color of the Razer Blade 15 Advanced Mid 2021.

Setting the static effect seems to ignore the color setting passed to it and always yields the default pink color, but using `breath_single` via the Python library client has the effect of a static backlight and does respect the color setting. Additionally, closing the lid or suspending the laptop reverts the color back to the default pink as well. Using the matrix to set the colors respects the colors and seems to survive closing the lid and suspension.

z3ntu, 2022-07-24:
@j-maynard @julian9499 Do you also see this issue on your device?

kkraus14, 2022-12-23:
not stale

kkraus14, 2023-05-10:
Not stale

z3ntu, 2023-05-10:
Unfortunately I can't help much here apart from suggesting you to 1. look for firmware updates for the device, and 2. try different parameters to see how the behavior changes. Without access to the device there's little possibility to debug here.

kkraus14, 2023-05-10:
@z3ntu I have no expectations here. I just don't want the bot to close the issue in case someone else stumbles onto this.