openrazer/openrazer #1860: 1532:010D no longer recognized

markaacosta, 2022-07-25:
### Distribution

Pop!_OS 22.04 LTS x86_64

### Support

- [YES ] Have you tried rebooting?
- [YES ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

For almost a year, I've been using OpenRazer with my Blackwidow Ultimate 2012 (1532:010D). It's allowed me to get access to my macro keys and map them to volume, other common functionality, etc. Recently, it stopped working. I start the daemon, but the macros aren't recognized; in the Polychromatic frontend, it says 'Unrecognized device'. Pop!_OS allows me to boot the old / previous kernel, so I did, and it started working again. This led me to assume I had corrupted some system file, and that a clean install would fix things; however, I've performed a clean OS install, and the problem is still here. I don't know how the kernel relates to the problem. I do have some other bits of info that may be relevant, and I can provide any logs or other files you need. Perhaps something there will help us determine how to resolve, or at least identify what's happening.

Thanks, Mark

lah7, 2022-07-25:
Please do provide the logs and output, then we'll be able to identify what has changed, or what went wrong during the kernel update. OpenRazer is a DKMS module (driver) that gets built when the kernel is installed. Possibly, it wasn't rebuilt during the upgrade.

[The output of 2 and 3 from the **troubleshooting guide**](https://github.com/openrazer/openrazer/wiki/Troubleshooting#2-kernel-modules-are-they-built-properly) will be of interest:

    sudo dkms install openrazer-driver/3.3.0
    sudo modprobe razerkbd

markaacosta, 2022-07-25:
Here they are - the second had no output

--

sudo dkms install openrazer-driver/3.3.0:

Module openrazer-driver/3.3.0 already installed on kernel 5.18.10-76051810-generic (x86_64).

--

sudo modprobe razerkbd:

--

lah7, 2022-07-25:
Thanks, that usually suggests the driver is loaded. The next point of failure might be the daemon or permissions.

Presumably, your user is in the `plugdev` group, check with:

    groups

If `plugdev` is not listed, run this, log out and log in again:

    sudo gpasswd -a $USER plugdev

---

Let's check the daemon: Stop it (`-s`), start again in the foreground (`-F`) and paste the output here:

```
openrazer-daemon -s
openrazer-daemon -F
```

_(Tip: When pasting here, surround the output with ``` )_

markaacosta, 2022-07-25:
'plugdev' is present with 'groups' invocation. here's the output from the two daemon commands:

```
openrazer-daemon -s
Module psutil is missing, not checking for process name
Process 3756 stopped
```

and...

```
openrazer-daemon -F
Starting daemon.
2022-07-25 01:54:23 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 67314
2022-07-25 01:54:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-07-25 01:54:23 | razer                          | INFO     | Serving DBus
```

lah7, 2022-07-25:
How are the permissions looking in the sysfs driver?

    ls -al /sys/bus/hid/drivers/razer*/*1532*/

This should output something like:

```
'/sys/bus/hid/drivers/razermouse/0003:1532:0046.000A/':
total 0
drwxr-xr-x 5 root plugdev    0 Jul 25 07:47 .
drwxr-xr-x 5 root root       0 Jul 25 07:47 ..
-r--r--r-- 1 root plugdev 4096 Jul 25 07:47 country
-rw-rw---- 1 root plugdev 4096 Jul 25 07:47 device_mode
-r--r----- 1 root plugdev 4096 Jul 25 07:47 device_serial
-r--r----- 1 root plugdev 4096 Jul 25 07:47 device_type
-rw-rw---- 1 root plugdev 4096 Jul 25 08:36 dpi
```

If they say `root root`, that would mean udev isn't setting the permissions as expected. Usually, replugging the device would do the trick.

markaacosta, 2022-07-25:
this globbing gives the output a few times; in all cases, 'root root' permission / group

```
/sys/bus/hid/drivers/razer/0003:1532:010D.000E/:
total 0
drwxr-xr-x 5 root root    0 Jul 25 02:19 .
drwxr-xr-x 5 root root    0 Jul 25 02:19 ..
-r--r--r-- 1 root root 4096 Jul 25 02:19 country
lrwxrwxrwx 1 root root    0 Jul 25 02:19 driver -> ../../../../../../../bus/hid/drivers/razer
drwxr-xr-x 3 root root    0 Jul 25 02:19 hidraw
drwxr-xr-x 3 root root    0 Jul 25 02:19 input
-r--r--r-- 1 root root 4096 Jul 25 02:19 modalias
drwxr-xr-x 2 root root    0 Jul 25 02:19 power
-r--r--r-- 1 root root 4096 Jul 25 02:19 report_descriptor
lrwxrwxrwx 1 root root    0 Jul 25 02:19 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Jul 25 02:19 uevent

/sys/bus/hid/drivers/razer/0003:1532:010D.000F/:
total 0
drwxr-xr-x 5 root root    0 Jul 25 02:19 .
drwxr-xr-x 5 root root    0 Jul 25 02:19 ..
-r--r--r-- 1 root root 4096 Jul 25 02:19 country
lrwxrwxrwx 1 root root    0 Jul 25 02:19 driver -> ../../../../../../../bus/hid/drivers/razer
drwxr-xr-x 3 root root    0 Jul 25 02:19 hidraw
drwxr-xr-x 3 root root    0 Jul 25 02:19 input
-r--r--r-- 1 root root 4096 Jul 25 02:19 modalias
drwxr-xr-x 2 root root    0 Jul 25 02:19 power
-r--r--r-- 1 root root 4096 Jul 25 02:19 report_descriptor
lrwxrwxrwx 1 root root    0 Jul 25 02:19 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Jul 25 02:19 uevent

/sys/bus/hid/drivers/razer/0003:1532:010D.0010/:
total 0
drwxr-xr-x 5 root root    0 Jul 25 02:19 .
drwxr-xr-x 5 root root    0 Jul 25 02:19 ..
-r--r--r-- 1 root root 4096 Jul 25 02:19 country
lrwxrwxrwx 1 root root    0 Jul 25 02:19 driver -> ../../../../../../../bus/hid/drivers/razer
drwxr-xr-x 3 root root    0 Jul 25 02:19 hidraw
drwxr-xr-x 3 root root    0 Jul 25 02:19 input
-r--r--r-- 1 root root 4096 Jul 25 02:19 modalias
drwxr-xr-x 2 root root    0 Jul 25 02:19 power
-r--r--r-- 1 root root 4096 Jul 25 02:19 report_descriptor
lrwxrwxrwx 1 root root    0 Jul 25 02:19 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Jul 25 02:19 uevent

```

markaacosta, 2022-07-25:
i've unplugged / replugged a few times and checked with each, but the 'root root' remains

markaacosta, 2022-07-25:
is this significant? when I used it (without issues) on my laptop, I think the symbolic links were in 'razerkbd', not 'razer' as I am seeing on my main machine

```
marka@pop-os:/sys/bus/hid/drivers$ ls -l **razer**
total 0
lrwxrwxrwx 1 root root    0 Jul 25 02:19 0003:1532:010D.000E -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:010D.000E
lrwxrwxrwx 1 root root    0 Jul 25 02:19 0003:1532:010D.000F -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:010D.000F
lrwxrwxrwx 1 root root    0 Jul 25 02:19 0003:1532:010D.0010 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:010D.0010
--w------- 1 root root 4096 Jul 25 00:34 bind
lrwxrwxrwx 1 root root    0 Jul 24 15:47 module -> ../../../../module/hid_razer
--w------- 1 root root 4096 Jul 25 00:34 new_id
--w------- 1 root root 4096 Jul 24 15:46 uevent
--w------- 1 root root 4096 Jul 25 00:34 unbind
marka@pop-os:/sys/bus/hid/drivers$ ls -l **razerkbd/**
total 0
--w------- 1 root root 4096 Jul 25 00:34 bind
lrwxrwxrwx 1 root root    0 Jul 25 00:34 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Jul 25 00:34 new_id
--w------- 1 root root 4096 Jul 24 15:46 uevent
--w------- 1 root root 4096 Jul 25 00:34 unbind
marka@pop-os:/sys/bus/hid/drivers$ 

```

z3ntu, 2022-07-25:
Uh yeah, for sure it shouldn't be called "razer" but "razerkbd", "razermouse" etc. What dkms version do you both have?

markaacosta, 2022-07-25:
dkms-2.8.7

markaacosta, 2022-07-25:
not to muddy the waters - i do also have a razer gaming headset that's usually plugged in. it's at least tangentially related, could be a factor so i thought i would mention

lah7, 2022-07-25:
Not sure what this `razer` driver is. That's probably why our `razerkbd` sysfs isn't showing what is expected, for example you might see:

```
-rw-rw---- 1 root plugdev 4096 Jul 25 18:01 device_mode
-r--r----- 1 root plugdev 4096 Jul 25 18:00 device_serial
-r--r----- 1 root plugdev 4096 Jul 25 18:00 device_type
lrwxrwxrwx 1 root plugdev    0 Jul 25 18:00 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Jul 25 18:00 kbd_layout
-rw-rw---- 1 root plugdev 4096 Jul 25 18:00 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Jul 25 18:00 macro_led_state
-rw-rw---- 1 root plugdev 4096 Jul 25 18:01 matrix_brightness
--w--w---- 1 root plugdev 4096 Jul 25 18:00 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Jul 25 18:00 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Jul 25 18:00 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Jul 25 18:00 matrix_effect_none
--w--w---- 1 root plugdev 4096 Jul 25 18:00 matrix_effect_reactive
```

What is the headset (`lsusb | grep 1532`)? Do you have another utility installed for managing those Razere devices? 

This command will let us know what other DKMS modules are installed:

    dkms status

lah7, 2022-07-25:
It's very likely the new Linux kernel is causing this, as [they've added a new `razer` driver](https://github.com/torvalds/linux/commit/047b6188b66e42513a2b0d36244f03d06f882e59) which involves your device:
```
RAZER_BLACKWIDOW_ULTIMATE   0x010D  <--
RAZER_BLACKWIDOW            0x010e
RAZER_BLACKWIDOW_CLASSIC    0x011b
```

There is a script at `/usr/lib/udev/razer_mount` that usually expects to unbind the device from `hid-generic`, but in this case, it needs to be `razer`. [I've made a patch](https://github.com/lah7/openrazer/commit/f5c4d22c4bd7d5215965dd8fbd0118bd3ad657b3 ), so let's try that: 

```bash
wget https://raw.githubusercontent.com/lah7/openrazer/f5c4d22c4bd7d5215965dd8fbd0118bd3ad657b3/install_files/udev/razer_mount > razer_mount
sudo mv razer_mount /usr/lib/udev/razer_mount
```

Then replug all your Razer devices (or just reboot the system) and let us know if all the devices are recognised. 🤞🏼

markaacosta, 2022-07-25:
the 'wget' pulls two files, 'razer_mount' and 'razer_mount.1'. 'razer_mount' is empty, so I delete it, rename 'razer_mount.1' to 'razer_mount', then run the other two lines to move it to the right spot and update the permissions.

However, I can confirm that all three of my razer devices are plugged in, and the keyboard (i.e. the one I use OpenRazer with) is working again! Thanks very much for the help - I am glad I hadn't simply missed something in the troubleshooting information! Regards, Mark

markaacosta, 2022-07-25:
And I can close the issue if you like; I'm not familiar with protocol / etiquette for the OpenRazer repository

z3ntu, 2022-07-25:
Just because you fixed it manually using the above doesn't mean it's fixed for everyone ;) keep it open

PR #1861 (2022-07-30): udev: Check binding for 'razer' driver in addition to 'hid-generic'
Fixes #1860.

[The upstream Linux kernel introduced a `razer` driver](https://github.com/torvalds/linux/commit/047b6188b66e42513a2b0d36244f03d06f882e59) for a handful of older Razer devices. This commit will check and unbind from that driver in addition to `hid-generic`, allowing these devices to continue using the OpenRazer driver.
commit 848408a7 (2022-07-30): udev: Check 'razer' as well as 'hid-generic'

Fixes #1860.

The upstream Linux kernel introduced a "razer" driver for a handful of
older Razer devices. This commit will check and unbind from that driver
in addition to hid-generic, so these devices use the OpenRazer driver.

https://github.com/torvalds/linux/commit/047b6188b66e42513a2b0d36244f03d06f882e59
closed 2022-07-30 by z3ntu