openrazer/openrazer #1866: razerkbd: module verification failed

serenegrace, 2022-07-30:
### Distribution

Huntsman Elite, Ubuntu 22.04

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I've tried uninstalling and reinstalling several times now, including purging everything.

`lsusb | grep 1532`
does not find the keyboard

the kernal modules, as per `sudo dkms install openrazer-driver/$version` (with $verion being 3.3.0), indicates it is already installed

`sudo modprobe razerkbd` returns nothing, as it is supposed to

I have added myself to the plugdev group

`lsmod | grep razer`
razerkbd               86016  0
hid                   147456  3 usbhid,razerkbd,hid_generic

`sudo dmesg | grep razer`
[  307.325908] razerkbd: module verification failed: signature and/or required key missing - tainting kernel

`sudo journalctl -b | grep -i razer_mount` returns nothing

`cat ~/.local/share/openrazer/logs/razer.log`

2022-07-29 21:25:02 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 10940
2022-07-29 21:25:02 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-07-29 21:25:02 | razer                          | INFO     | Serving DBus
2022-07-29 21:33:45 | razer                          | INFO     | Stopping daemon on signal 15
2022-07-29 21:34:53 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 2218
2022-07-29 21:34:53 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-07-29 21:34:53 | razer                          | INFO     | Serving DBus
2022-07-29 22:10:59 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 2158
2022-07-29 22:10:59 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-07-29 22:10:59 | razer                          | INFO     | Serving DBus
2022-07-29 22:23:03 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 2177
2022-07-29 22:23:03 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-07-29 22:23:03 | razer                          | INFO     | Serving DBus

`ls -al /sys/bus/hid/drivers/razer*/*1532*/`
ls: cannot access '/sys/bus/hid/drivers/razer*/*1532*/': No such file or directory

Having attempted to find a solution, i found https://github.com/openrazer/openrazer/issues/574 and https://github.com/polychromatic/polychromatic/issues/146#issuecomment-379465409
. Having followed both, neither worked. 

Help appreciated

z3ntu, 2022-07-30:
If the lsusb command doesn't show anything then the kernel doesn't think it's connected. If you're 100% sure it's connected (and working as in you can type stuff) please send lsusb output without the grep

serenegrace, 2022-07-30:
It is 100% connected, but it is not working (does not type anything).

lah7, 2022-07-30:
Try another USB port or hub. Sounds like a hardware fault if the kernel doesn't list the device in `lsusb`. Maybe the keyboard is stuck in some kind of factory/demo mode?

serenegrace, 2022-07-30:
I've tried every usb port and hub, and confirmed they work (with other keyboards and other usb devices). I'll look into factory mode/demo thing.

z3ntu, 2022-07-30:
@lah7 If you mean the Razer bootloader mode, then it'll still show up as USB device. 

@serenegrace Did this happen before installing OpenRazer? I pretty much doubt it's at fault here. Did you try it on another computer?

serenegrace, 2022-07-30:
Trying now on another computer. 

I'm not sure if this happened before or after, as I preemptively installed openrazer. Based on what has been mentioned, it seems likely that this was in fact something unrelated. Sorry for having bothered you.

serenegrace, 2022-07-30:
Issue confirmed to be unrelated. Sorry for the confusion. (For what it is worth, you did help me confirm this was not the issue which was very helpful).

closed 2022-07-30 by serenegrace