openrazer/openrazer #1869: Support for Razer Hyperpolling Wireless Dongle

ginzberg, 2022-07-31:
### Device Info

* Name: Razer Hyperpolling Wireless Dongle
* Product Number: RC30-04410100-R3M1
* Model Number: HSD01 (NOTE: On box it says "HDS01", On the bottom it says "HSD01". I find it easier to believe that this is a misprinting on the packaging rather than a misprinting on the engraved text on the bottom of the dongle.)

### Device Information
```
Bus 001 Device 004: ID 1532:00b3 Razer USA, Ltd Razer HyperPolling Wireless Dongle
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b3
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer HyperPolling Wireless Dongle
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              2 Razer HyperPolling Wireless Dongle
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      0
      iInterface              2 Razer HyperPolling Wireless Dongle
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     260
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              2 Razer HyperPolling Wireless Dongle
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      63
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx    10 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-event-if01 -> ../event23
lrwxrwxrwx    10 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-event-mouse -> ../event18
lrwxrwxrwx    10 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-if01-event-kbd -> ../event21
lrwxrwxrwx    10 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-if01-event-mouse -> ../event22
lrwxrwxrwx     9 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-if01-mouse -> ../mouse4
lrwxrwxrwx    10 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-if02-event-kbd -> ../event24
lrwxrwxrwx     9 root 31 Jul 12:12  usb-Razer_Razer_HyperPolling_Wireless_Dongle-mouse -> ../mouse3
```

### Packet Captures
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_hyperpolling_led_set_to_battery_status.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_hyperpolling_led_set_to_battery_warning_only.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_hyperpolling_led_set_to_connection_status.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_viper_v2_pro_set_polling_4000.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_viper_v2_pro_set_polling_2000.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_viper_v2_pro_set_polling_1000.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_viper_v2_pro_set_polling_500.pcapng
https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/wireshark_razer_viper_v2_pro_set_polling_125.pcapng

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
[image]
[image]
[image]
[image]

ginzberg, 2022-07-31:
Some additional context:
This product is a wireless dongle that will be compatible with  multiple devices. To date, the first product is the Viper V2 Pro.
There is a pairing workflow for this product that I did not log for this submission (though I can provide if there is interest). 
The pairing workflow is necessary to setup the dongle to work with your mouse. Once paired, however, the original wireless dongle that originally came with the mouse is no longer required. That said, they can both co-exist on the system. If both are present, the hyperpolling dongle is prioritized. I can confirm this is happening in Linux using [Input Polling Test](https://github.com/ad8e/input-polling-test/) to confirm 4khz and 1khz polling depending on which combination of dongles are plugged in. 

Happy to do additional tests if requested.

ginzberg, 2022-09-17:
I have now gotten captures for both pairing and unpairing. 
[01-pairing_clicking_on_pairing_utility.pcapng](https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/01-pairing_clicked_on_pairing_utility.pcapng)
[02-pairing_pairing_successful.pcapng](https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/02-pairing_pairing_successful.pcapng)
[03-unpair_clicked_on_dongle.pcapng](https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/03-unpair_clicked_on_dongle.pcapng)
[04-unpair_select_unpair.pcapng](https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/04-unpair_select_unpair.pcapng)
[05-unpair_confirm_unpair.pcapng](https://github.com/ginzberg/razer-hyperpolling-captures/blob/main/05-unpair_confirm_unpair.pcapng)

BLooD-DoG, 2022-09-23:
Thanks to ginzberg's captures, I have a branch @ https://github.com/BLooD-DoG/openrazer/tree/Hyperpolling_Wireless_Dongle with some preliminary support. Some notes:

1- The HyperPolling Wireless Dongle allows for 2KHz and 4KHz polling for certain mice (at this time, the Viper V2 Pro and the DeathAdder V3 Pro, with probably more to come). This is in contrast to the max 1KHz polling in wired mode, or in wireless mode with the standard dongle included with the mouse. This is the only extra mouse feature versus wired or "normal" wireless capabilities out of the box. There is also a feature to configure the LED on the dongle, see below.

2- In Synapse, the dongle shows up as both the Mouse it's paired with (Viper V2 Pro or DA V3 Pro), and also as an Accessory. The Mouse tab is identical to what it would be without this dongle (wired or normal wireless), but the polling options now include 2Khz and 4KHz. The Accessory tab gives you the Pair/Unpair options, as well as the option to configure the LED on the dongle to 3 different settings. Synapse is just showing it as 2 separate devices cosmetically even though it's a single USB device that is connected. Doing something similar in openrazer seems more complicated than it's worth (supporting 2 devices with the same USB ID), so I just added the extra LED functionality under the "Mouse" device type.

3- As of now I haven't added the Pair/Unpair functionality (not sure if I will get to it), but I did add the extra polling rate options, and the LED configuration options. To configure the LED, do something like `echo 1 > /sys/module/razermouse/drivers/hid:razermouse/*/hyperpolling_wireless_dongle_indicator_led_mode` where 1 can be 2 or 3. 1 = Connection Status, 2 = Battery Status, 3 = Battery Warning

4- Getting the LED status after it's set doesn't seem to work, although I am calling `get_razer_report(0x07, 0x90, 0x01);`, which is what shows up in the capture when you run Synapse and go to the dongle Accessory tab. It's possible there is no way to get the status, and Synapse just shows what was last set (and assumes it wasn't changed elsewhere). This is cosmetic anyway since changing the setting will always work.

5- The image of the device is hardcoded to a DA V3 Pro, but ideally it would dynamically use the image of the type of mouse it's paired with, as Synapse does.

The branch has a couple of things not strictly related to supporting this dongle:

6- It builds on top of the PR I made to add support for the DeathAdder V3 Pro.

7- In driver/razerchrommacommon.c and scripts/razer_mouse/driver/set_mouse_dpi.py
, the DPI was clamped to 26000, I assume because that was the highest DPI when support for the Basilisk V3 was added. However, the Viper V2 Pro and DA V3 Pro (and probably others) support 30000 DPI, so I increased the limit.

8- In driver/razerchromacommon.c, razer_chroma_misc_get_polling_rate2(), there is a TODO comment that Razer sends each request twice, once with 0x00 and once with 0x01. I observed this as well for my DA V3 Pro using this dongle. So I added an argument to the method and now call it twice in razermouse_driver.c for the only 2 mice that call this method (the Viper 8K and the HyperPolling Wireless Dongle). There is also a keyboard that calls this method in driver/razerkbd_driver.c (Huntsman V2) but I haven't changed it to call the method twice.

Someone with a better understanding of openrazer can probably clean this up quite a bit, but it works well enough for my needs.

BLooD-DoG, 2022-09-24:
In case it helps, after the dongle was paired I took a Wireshark capture of a clean reboot into Windows + Login + Run Synapse + go to dongle Accessory tab. It helps to filter with `usb.data_fragment > 0`, and the above steps are labeled as Packet Comments ("End of boot", "End of login", "End of launch Synapse and go to Dongle page").
[Wireshark_boot_and_synapse.pcapng.zip](https://github.com/openrazer/openrazer/files/9638408/Wireshark_boot_and_synapse.pcapng.zip)

BLooD-DoG, 2022-10-06:
I ended up adding support for pairing and unpairing in the branch @ https://github.com/openrazer/openrazer/compare/master...BLooD-DoG:openrazer:Hyperpolling_Wireless_Dongle , I will submit a Pull Request as everything seems to be working fine.

Instructions:
1- For setting the indicator LED mode, do `echo 1 > /sys/module/razermouse/drivers/hid:razermouse/*/hyperpolling_wireless_dongle_indicator_led_mode` where 1 = Connection Status, 2 = Battery Status, 3 = Battery Warning

2- To pair the dongle with a mouse, do  `echo b7 > /sys/module/razermouse/drivers/hid:razermouse/*/hyperpolling_wireless_dongle_pair` where b7 = DeathAdder V3 Pro, a6 = Viper V2 Pro, etc. This should work for any Product ID that is supported by the dongle. Note that Synapse doesn't ask for the PID or mouse type, I believe it just tries to pair with the mouse that was previously connected wirelessly, or that is currently connected (wired) during the time of pairing. This is just a frontend convenience. If you enter an invalid PID, or try to pair when a mouse is already paired, the firmware will give up after 1 minute and everything will be back to normal. If you were already paired prior to trying to break it, you will remain paired after the 1 minute and everything will continue to work. Synapse tries to shield you from these error states (e.g. by not letting you Pair if the dongle is already paired), but again this is just a front end convenience.

3- To unpair the dongle from a mouse, do  `echo b7 > /sys/module/razermouse/drivers/hid:razermouse/*/hyperpolling_wireless_dongle_unpair`. As above for pairing, replace b7 by the PID of the mouse that is paired. Also as above, no harm in trying to unpair if nothing is paired, or trying to unpair with an invalid PID.

The only TODO I can think of is to dynamically change the image of the dongle to the image of the mouse that is paired, as Synapse does, but this is a minor cosmetic convenience.

The branch has a couple of things not strictly related to supporting this dongle:

4- It builds on top of the PR I made to add support for the DeathAdder V3 Pro, since that is the mouse I have and was testing with.

5- In `driver/razerchrommacommon.c` and `scripts/razer_mouse/driver/set_mouse_dpi.py`
, the DPI was clamped to 26000, I assume because that was the highest DPI when support for the Basilisk V3 was added. However, the Viper V2 Pro and DA V3 Pro (and probably others) support 30000 DPI, so I increased the limit.

6- In `driver/razerchromacommon.c`, `razer_chroma_misc_get_polling_rate2()`, there was a TODO comment that Razer sends each request twice, once with 0x00 and once with 0x01. I observed this as well for my DA V3 Pro using this dongle. So I added an argument to the method and now call it twice in razermouse_driver.c for the only 2 mice that call this method (the Viper 8K and the HyperPolling Wireless Dongle). There is also a keyboard that calls this method in driver/razerkbd_driver.c (Huntsman V2) but I haven't changed it to call the method twice.

ginzberg, 2022-10-06:
I have been able to perform all actions on my Viper V2 Pro + Hyperpolling 4khz dongle with @BLooD-DoG 's branch, including:
updating led mode
unpairing
pairing

z3ntu, 2022-10-15:
> Model Number: HSD01 (NOTE: On box it says "HDS01", On the bottom it says "HSD01". I find it easier to believe that this is a misprinting on the packaging rather than a misprinting on the engraved text on the bottom of the dongle.)

FCC agrees with you :wink: https://fccid.io/RWO-HSD01

closed 2023-02-11 by z3ntu

R4id0ss, 2024-04-03:
any update for 8khz polling?

Can I help somehow?
I have a Basilisk V3 Pro with the hyperpolling (until now it works at 4khz)

z3ntu, 2024-04-04:
@R4id0ss I guess you mean #2189 ?

dm17, 2024-09-11:
Using `evhz` I can see the mouse varies between 31hz and 4081Hz while moving it around. Unfortunately it seems to stutter when moving it quickly. I paired it using windows to the hyperpolling dongle. Has anyone else dealth with this stuttering? Is there any hope or should I move back to the dongle it came with? I didn't notice hyperpolling dongle causing stutter in windows.

BLooD-DoG, 2024-09-11:
I haven't had an issue, and neither have the (admittedly few) people I know who've tried it. I'm not an expert in this but I would suggest trying a higher DPI in conjunction with the higher refresh rates and see if that helps.

dm17, 2024-09-22:
Anyone get the settings to stick between reboots and mouse disconnection/reconnection? The polling rate resets.

BLooD-DoG, 2024-09-23:
Mine sticks pretty much all the time, including when rebooting between Windows and Linux and unplugging replugging. However, I have noticed that sometimes (far from always) it seems to reset to 500Hz when booting into linux, at least according to polychromatic. However the last time this happened, I ran evhz and it was still running at 2000Hz, so I wonder if that's just a cosmetic mis-reporting.

dm17, 2024-09-24:
Is there nowhere the settings can be saved? I don't know how razergenie works, but it is the only way I see to turn the polling rate up. RazerGenie doesn't seem to have any support for the top razer mice, but it can turn up the hyperpolling dongle (which is what I'm doing). I'd prefer to uninstall all of the useless razer software and just save the polling rate somewhere.