openrazer/openrazer #1871: How to turn off Razer logo in the back of laptop?

lorenzo-ulter, 2022-08-04:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I see no option in polychromatic to control my Razer Blade 17 pro (2020) logo lights. But googling around I see some other people with different models saying there is a button to toggle it on and off, so I was dondering if my laptop didn't have support for that or maybe i'm doing something wrong? Thanks in advance

lah7, 2022-08-06:
If your device is definitely the `1532:0279`, then it should support the logo toggle:

https://github.com/openrazer/openrazer/blob/80f5a3b6cf04c8bc94ecf9971d47e2a038666bbf/daemon/openrazer_daemon/hardware/keyboards.py#L1517

Polychromatic should show a brightness checkbox under "Logo" or "Laptop Lid". If that's not present, please could you let me know the version of the app and a screenshot of what options you see?

lorenzo-ulter, 2022-08-06:
[image]

These are the options that I see, the model you mention 1532:0279 is the 2021 model, mine is the 2020 model (I don't know how to find the equivalent number for mine)

lah7, 2022-08-06:
Ahh okay, yours is `1532:0256` (can also find by clicking **Device Info**). It seems from #1376 that the [original contributor had a broken logo](https://github.com/openrazer/openrazer/pull/1376#issuecomment-950140258), so it wasn't added.

https://github.com/openrazer/openrazer/blob/80f5a3b6cf04c8bc94ecf9971d47e2a038666bbf/daemon/openrazer_daemon/hardware/keyboards.py#L1414-L1426

If you're lucky, it might be a simple patch to add logo support. I'm unsure if something else needs tweaking in the driver too should this not work. We can give it a shot:

1. Open `/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py` in text editor as root.
2. Find `RazerBladeProEarly2020` and add at the end of the `METHODS` line:
    ```python
    'get_logo_active', 'set_logo_active',
    ```

    Like this - syntax is important:

    ```diff
         MATRIX_DIMS = [6, 16] 
    -     METHODS = ['get_device_type_keyboard', 'set_wave_effect', 'set_static_effect', 'set_spectrum_effect', 
    +     METHODS = ['get_device_type_keyboard', 'set_wave_effect', 'set_static_effect', 'set_spectrum_effect', 'get_logo_active', 'set_logo_active',
                    'set_reactive_effect', 'set_none_effect', 'set_custom_effect', 'set_key_row', 
    ```
3. Restart the OpenRazer daemon. 
     In Polychromatic, you can do this under **Tools** → OpenRazer → Restart Daemon.

Let us know if this works 🤞🏼

lorenzo-ulter, 2022-08-06:
Ok I have added the extra methods to the file, screenshot below

[image]

But now polichromatic stays like this

[image]

lah7, 2022-08-07:
Try running both the OpenRazer daemon and Polychromatic in two terminals, this should reveal the cause of the hang.

    openrazer-daemon -rFv
    polychromatic-controller -v

(Copy & paste the terminal output here and press CTRL+E to wrap them in `code` ``` tags)

lorenzo-ulter, 2022-08-07:
For polichromatic I get this
```
lorenzo@Blade:~$ polychromatic-controller -v
Initialising application...
Initialising window...
Initialising troubleshooters...
Opening tab: 0
Initialising backends...
[openrazer] Initialising Device Manager...
QObject::setParent: Cannot set parent, new parent is in a different thread
ERROR:dbus.proxies:Introspect error on :1.361:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Backends loaded in 50.364s.
QThread::start: Thread termination error: No such process
Opening tab: 0
[openrazer] Initialising Device Manager...
ERROR:dbus.proxies:Introspect error on :1.361:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeexpired, or the network connection was broken.

```

For Daemon I get this

```
lorenzo@Blade:~$ openrazer-daemon -rFv
Process 22162 stopped
Starting daemon.
2022-08-07 10:28:12 | razer                          | INFO     | Initialising Daemon (v3.4.0). Pid: 22208
2022-08-07 10:28:12 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2022-08-07 10:28:12 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
[... cut ...]
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 223, in set_logo_active
    with open(driver_path, 'w') as driver_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0256.0009/logo_led_state'

2022-08-07 10:28:14 | razer.device0                  | DEBUG    | DBus call get_battery
2022-08-07 10:28:14 | razer.device0.batterynotifier  | DEBUG    | Razer DeathAdder V2 Pro Wireless Battery at 0%
       
```

lorenzo-ulter, 2022-08-07:
I think I took too long to copy paste and it tried to run again so everything is doubled I think, however I think this is what we're interested in:

```
2022-08-07 10:28:14 | razer.device1                  | DEBUG    | DBus call set_logo_active
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 117, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 491, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 56, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 26, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 1178, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 315, in __init__
    active_func(self.zone[i]["active"])
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 223, in set_logo_active
    with open(driver_path, 'w') as driver_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0256.0009/logo_led_state'
```

Etaash-mathamsetty, 2022-08-09:
> I think I took too long to copy paste and it tried to run again so everything is doubled I think, however I think this is what we're interested in:
>
> ```
> 2022-08-07 10:28:14 | [...]

seems like some kind of permission issue, if a reboot doesn't fix it, maybe try checking what permissions you need to write to that file

lah7, 2022-08-09:
Could be a permission issue as @Etaash-mathamsetty points out. But thinking about it, I don't think the file exists:

    /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0256.0009/logo_led_state

We'll need to add a couple of lines to the driver. I've [created a commit](https://github.com/lah7/openrazer/commit/2fea9ce40d9e98272ccd2919f839abf282149238) in my fork of the changes. Please test:

```bash
git clone https://github.com/lah7/openrazer -b add_logo_RazerBladeProEarly2020
cd openrazer
./scripts/build_debs.sh
sudo apt install ./dist/*.deb
```

Might need to `sudo apt install git build-essential` if the build fails. Also, be sure to reboot (to fully load the updated driver) before trying Polychromatic again. Hopefully it works this time 🤞🏼

lorenzo-ulter, 2022-08-19:
I installed build-essentials, however running builds_debs.sh fails with the following error:

`lorenzo@Blade:~$ cd openrazer
lorenzo@Blade:~/openrazer$ ./scripts/build_debs.sh
Temp DIR: /tmp/tmp.ZKiZky5UB9_deb_build_tmp
Copying source to temporary build directory
Building deb packages

Failed to generate deb files. Check /tmp/tmp.ZKiZky5UB9_deb_build_tmp/build.log for more details
`

I opened the file in question to see what the error is, I get the following:

`dpkg-buildpackage: info: source package openrazer
dpkg-buildpackage: info: source version 3.4.0-0
dpkg-buildpackage: info: source distribution jammy
dpkg-buildpackage: info: source changed by Luca Weiss <luca@z3ntu.xyz>
 dpkg-source --before-build .
dpkg-buildpackage: info: host architecture amd64
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9) dh-python
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)`

So I ran sudo apt-get install debhelper and same with dh-python and then I was able to run everything else just fine. Rebooting now to see if it all works, just wanted to post this prior to rebooting so the record is there

lorenzo-ulter, 2022-08-19:
> Could be a permission issue as @Etaash-mathamsetty points out. But thinking about it, I don't think the file exists:
> 
> ```
> /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0256.0009/logo_led_state
> ```
> 
> We'll need to add a couple of lines to the driver. I've [created a commit](https://github.com/lah7/openrazer/commit/2fea9ce40d9e98272ccd2919f839abf282149238) in my fork of the changes. Please test:
> 
> ```shell
> git clone https://github.com/lah7/openrazer -b add_logo_RazerBladeProEarly2020
> cd openrazer
> ./scripts/build_debs.sh
> sudo apt install ./dist/*.deb
> ```
> 
> Might need to `sudo apt install git build-essential` if the build fails. Also, be sure to reboot (to fully load the updated driver) before trying Polychromatic again. Hopefully it works this time 🤞🏼

Alright after successfully building and installing this update, polychromatic still hangs, I get this error on the terminal

`lorenzo@Blade:~$ polychromatic-controller -v
Initialising application...
Initialising window...
Initialising troubleshooters...
Opening tab: 0

Initialising backends...
[openrazer] Initialising Device Manager...
QObject::setParent: Cannot set parent, new parent is in a different thread
ERROR:dbus.proxies:Introspect error on :1.167:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Backends loaded in 50.342s.
QThread::start: Thread termination error: No such process
`

and the final parts of the responde to openrazer-daemon is:

`2022-08-18 22:33:29 | razer.device1                  | DEBUG    | Sending observer message: ('effect', RazerBladeProEarly2020:BY2027A97402833, 'setBrightness', 23)
2022-08-18 22:33:29 | razer.device1                  | DEBUG    | DBus call set_logo_active
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 117, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 491, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 56, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 26, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 1178, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 315, in __init__
    active_func(self.zone[i]["active"])
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 223, in set_logo_active
    with open(driver_path, 'w') as driver_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0256.0009/logo_led_state'
`

closed 2023-01-17 by github-actions[bot]