openrazer/openrazer #1872: Adding support for new device

lcuminato, 2022-08-05:
Hi there,

I would like to add support for my desktop case, Bus 001 Device 007: ID 1532:0f17 Razer USA, Ltd Razer Tomahawk Chassis, but where can I find proper documentation about my particular device ?

lah7, 2022-08-06:
Hi. Please close this issue and create a new one using the [New Device Support](https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&template=new-device-support.md&title=Support+for+Razer+Tomahawk+Chassis) template. 

While one could guess what protocols your device speaks, it's more helpful to provide a copy of packet captures - which requires a temporary install of Windows and Razer Synapse. 

If you'd like to implement it yourself (pull requests encouraged!), take a look at the [reverse engineering guide](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) and/or [look at commits implementing other devices](https://github.com/openrazer/openrazer/search?q=Add+support+for&type=commits) to know which lines you need to add and modify to make it work for your device. You can test your changes by [building a package.](https://github.com/openrazer/openrazer/wiki/Building-a-package)

lcuminato, 2022-08-06:
Thanks for the info. Yeah I will do it myself and PR my changes if they work. So basically we have to sniff the traffic between Windows and the USB device to determine the protocol ?? Isn't this project backed by Razer therefore I would imagine official support is provided ? Anyways, thanks again for the info.

lah7, 2022-08-07:
Having a copy of pcaps (packet captures) helps as a reference later in case new stuff comes up. Most Razer devices these days are software effect driven, but they might have their hardware effects still built into the firmware.  If you're lucky, this device will use the same command IDs as existing devices and just needs this device adding in a few places.

Sadly, OpenRazer has no affiliation with Razer Inc, it's entirely community supported.