openrazer/openrazer #1873: "daemon is not installed"

Douper2, 2022-08-05:
[image]
also this is happening on manjaro
ive installed razergenie from the flathub

lah7, 2022-08-06:
For Manjaro, have you installed `openrazer-meta` from the AUR?

Douper2, 2022-08-08:
> For Manjaro, have you installed `openrazer-meta` from the AUR?

no, i did not

z3ntu, 2022-08-08:
Please do so then, as mentioned in the error message in the screenshot you sent.

Douper2, 2022-08-08:
> Please do so then, as mentioned in the error message in the screenshot you sent.

ok, i will try to do that

Douper2, 2022-08-08:
> Please do so then, as mentioned in the error message in the screenshot you sent.

so i checked if i have it installed and i have it installed

z3ntu, 2022-08-09:
Did you check our Troubleshooting page already? https://github.com/openrazer/openrazer/wiki/Troubleshooting

Douper2, 2022-08-10:
> Did you check our Troubleshooting page already? https://github.com/openrazer/openrazer/wiki/Troubleshooting

hey, ive fixed the issue by using another windows pc. also thanks for all the things that you said.

closed 2022-08-10 by Douper2