openrazer/openrazer #1874: Razer Blade 14 (2021) Keys are swapped in the grid/matrix

BeanieBen9990, 2022-08-05:
### Distribution

Any distribution and most up to date version of openrazer

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Razer Blade 14 (2021) Keyboard grid does not match correctly with the layout of the keys. Specifically, if you open polychromatic and individually select certain coordinates of keys, the rightmost and lowest key on the keyboard, the right arrow key, is mapped to the lowest row, but the second to last column, and the key that should be the second last column in the bottom row, the down arrow key, is mapped to the last column. In other words, going by the grid, the Arrow Down key and the Arrow Right keys are switched. I've looked through the code and don't know if there's a way to switch the key mapping such that effects that rely on correct grid layout are seen as correct on the physical device.

Also, the device is laid out with 16 columns, but the first column (column 0) is not mapped to any keys at all

BeanieBen9990, 2022-08-05:
![Uploading 183146086...]
![Uploading 183146103...]
![Uploading 183146115...]

BeanieBen9990, 2022-08-05:
Here's some images to make the description more clear

[image]
[image]
[image]

lah7, 2022-08-06:
Is the graphic (not the grid) correct? If not, please raise the issue on Polychromatic's repository and highlight which keys are good (green) and which are wrong (red), then a copy can be created for this device. These graphical mappings are not provided by OpenRazer.

For the grid, it's not a "bug" as such, but rather how Razer maps onto the hardware: https://developer.razer.com/works-with-chroma/razer-chroma-led-profiles/  (there is/was a page that had device-specifics, can't find it right now)

It is a good point that custom effects using OpenRazer's API could suffer if it's expected that each column maps directly to the physical LED position. For say, this BlackWidow Chroma, the grid is pretty accurate. OpenRazer utilises hardware effects for things like Wave, so the device's firmware should handle them correctly.

z3ntu, 2022-10-15:
Closing as it's very likely a polychromatic issue, not an openrazer one.

closed 2022-10-15 by z3ntu