openrazer/openrazer #1875: Razer Mamba sometimes not moving but buttons worked.

Sebastiangperez, 2022-08-05:
### Distribution
Pop OS 21.10

Razer Mamba 

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Sometimes my mouse do not move at all , but this not happen very often maybe one in a month, i tried to unplug and plug in another usb port and the behavior is the same , any thought.

lah7, 2022-08-06:
Sounds like a similiar issue with my Razer Mamba Tournament Edition mouse. For me, it's extremely rare, but the cursor just stops moving, while click and scroll works. My workaround is to press the DPI buttons, and then it returns to normal.

I think it's a hardware quirk, like the optical sensor just had a snooze. 💤

Sebastiangperez, 2022-08-06:
I tried that but still doing the same stuff, maybe i will test it in another computer, sometimes do not happen at all for weeks but sometimes it's constantly, like today , 10 times today and counting 
There is some command that monitoring the mouse behavior , i mean , laser and stuff or do not exist such thing?

lah7, 2022-08-06:
I'm not aware of any monitoring tools that would be of any use to this situation. You could clean the mouse in case dust is interfering with the sensor. Dust caused problems with erratic scroll wheel for me.

Maybe check the logs after it happens (via the Terminal) and see if there's any messages related to USB ports or input:

    sudo dmesg
    sudo cat /var/log/Xorg.0.log

closed 2023-01-05 by github-actions[bot]