openrazer/openrazer #1879: PyPI parity

meeuw, 2022-08-09:
I'm trying to package openrazer for Fedora (to get it in the rpmfusion repository)

But, according to:

https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#_pypi_parity

> Every Python package in Fedora SHOULD also be available on [the Python Package Index](https://pypi.org/) (PyPI).

Not sure what the package names should be and how to set expectations on `pip install`.

meeuw, 2022-08-09:
I understand your worries that people might expect everything to work when they run `pip install openrazer`.

On the other hand, PyPI is full of packages which have dependencies outside of Python (OS specific, dependencies on specific kernel versions/features etc).

In my opinion it's perfectly fine to publish something on pypi which covers only 1% of the cases. I think it's important to set expectations (add a message to the description of the package) and give sane error messages (like openrazer already has).

Anyway, for Linux distributions it's important not to have conflicts in system-site-packages, please make sure openrazer owns the `openrazer` and `openrazer_daemon` packages in PyPI (either by publishing the packages or just block the names).

Do you need some help to get in touch with the current owner of openrazer on pypi?

z3ntu, 2022-08-09:
I've already tried years ago to get in contact with the person owning "openrazer" on PyPI but never got a response.

But still I don't see any added value for anyone publishing any part of openrazer on pypi (other than reserving the name), and it'll only make things worse when people install openrazer via package manager after `pip install --system openrazer` (or whatever) and get conflicts from their package manager then. OpenRazer won't be of any use without kernel modules plus daemon plus all the init files, like systemd unit file and udev rules etc.

meeuw, 2022-08-09:
Did you try to send an e-mail to him (e-mail address can be found from one of his commits)? @daniellawrence seems to have a nice @linux.com e-mail address...

Another value of publishing to pypi is (for Fedora) that the python rpm macros can be used (to automatically find the dependencies, get the sources, generate the files list etc).

It's already possible to run `pip install .` as root and do a lot of similar damage.

For debugging purposes one could build and load the kernel modules, install openrazer in a virtualenv from pypi and run it as root.

z3ntu, 2022-08-09:
> Did you try to send an e-mail to him (e-mail address can be found from one of his commits)? @daniellawrence seems to have a nice @linux.com e-mail address...

I don't recall and quickly looking through my inbox I didn't find anything but it's possible. But at least I haven't heard of anyone installing that package and expecting this openrazer to be installed so currently it's not an issue at all.

> Another value of publishing to pypi is (for Fedora) that the python rpm macros can be used (to automatically find the dependencies, get the sources, generate the files list etc).

I'm anyways maintaining this spec file https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec for Fedora, openSUSE and Mageia so in any case what you do, it still needs to be maintained for the other two distros with all the `if` statements in there (because why can't package names etc be consistent) so it doesn't really change anything for me.

> It's already possible to run pip install . as root and do a lot of similar damage.

If you want to break your system, you easily can ;) But given that pypi has a giant `pip install $pkgname` on the header it's a bit more likely users run that. We don't use Python easy installability or anything, it's really just an implementation detail.

> For debugging purposes one could build and load the kernel modules, install openrazer in a virtualenv from pypi and run it as root.

Or you just set PYTHONPATH to the sources and don't need to bother with a virtualenv at all. See e.g. https://gist.github.com/z3ntu/c5ca8f6040d8934fafc5e686b234890e

meeuw, 2022-08-09:
:v: I give up on trying to convince you to publish openrazer on pypi. It's not a big problem and the gains are low.

But please 🙏 : Get openrazer and openrazer_daemon claimed/blocked at PyPI.

Somewhat offtopic: I don't think the openrazer spec file will ever get approved in it's current form by rpmfusion. I think it would be awesome if people could just install openrazer from a (semi) official repository so I'm trying to achieve that.

daniellawrence, 2022-08-09:
deleted https://pypi.org/project/openrazer/ - let me know if anything else needs to happen re: namespace squatting.

z3ntu, 2023-01-08:
I've uploaded dummy packages for both `openrazer` and `openrazer_daemon` now in pypi, inspired by the recent pytorch news https://pytorch.org/blog/compromised-nightly-dependency/

At least noone should be able to claim and upload random stuff under these names now.

closed 2023-01-08 by z3ntu