openrazer/openrazer #1882: Sensitivity buttons snaps DPI back to 2600.

comand100vip, 2022-08-10:
### Distribution

Manjaro Linux 21.3.6

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have `Razer Basilisk` and I configured it's DPI levels as follows:
`800 1000 1200 1400 1600`
And it seems that it was applied, but after pressing the any of two sensitivity buttons located in the middle of the mouse snaps DPI back to it's default `2600` and won't change unless I manually do it from my computer.
Also I am using `Polychromatic` as a front-end.

Thanks in advice!

lah7, 2022-08-10:
It's expected behaviour at the moment. OpenRazer doesn't currently intercept the DPI buttons in software, so these buttons are stuck with the hardware's factory presets. 

Only a handful of mice have the capability of saving a custom range of "DPI stages" to the firmware. It seems a few more mice might actually have support this hardware feature (#1876)

comand100vip, 2022-08-10:
Okay thanks for the info.

closed 2022-08-10 by comand100vip