openrazer/openrazer #1885: Fedora Silverblue 36 - No devices detected

FabienOnGithub, 2022-08-12:
Hello !

I'm a total newbie in building packages but I finally managed to make openrazer work on Silverblue thanks to this [thread](https://github.com/openrazer/OBS-packaging/pull/8)

I have a huntsman v2 analog and a basilisk ultimate, here's my lsusb : 

`Bus 005 Device 002: ID 1532:0266 Razer USA, Ltd Razer Huntsman V2 Analog`
`Bus 001 Device 009: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock`
`Bus 001 Device 008: ID 1532:0088 Razer USA, Ltd Gaming Mouse [Basilisk Ultimate, Wireless, Receiver]`

And for the installation here's the step I followed in my terminal: 

`toolbox enter`
`sudo dnf install rpm-build rpmdevtools`
`rpmdev-setuptree`
`cd ~/rpmbuild/SPECS`
`curl -O https://raw.githubusercontent.com/kyokenn/OBS-packaging/master/openrazer-kmod-common.spec`
`curl -O https://raw.githubusercontent.com/kyokenn/OBS-packaging/master/openrazer-kmod.spec`
`curl -O https://raw.githubusercontent.com/kyokenn/OBS-packaging/master/openrazer.spec`
`spectool -g -R openrazer-kmod-common.spec`
`spectool -g -R openrazer-kmod.spec`
`spectool -g -R openrazer.spec`
`sudo dnf -y builddep openrazer-kmod-common.spec`
`sudo dnf -y builddep openrazer-kmod.spec`
`sudo dnf -y builddep openrazer.spec`
`rpmbuild -bb openrazer-kmod-common.spec`
`rpmbuild -bb openrazer-kmod.spec`
`rpmbuild -bb openrazer.spec`
`exit`
`cd ~/rpmbuild/RPMS/x86_64`
`rpm-ostree install *.rpm`
`cd ~/rpmbuild/RPMS/`
`rm (the dkms one, I don't remember the full name)`
`rpm-ostree install *.rpm`
`systemctl reboot`

After that I had a problem with the plugdev group so I did : 
`grep -E '^plugdev:' /usr/lib/group >> /etc/group`
and then
`usermod -aG plugdev <my user>`

Now i have this situation, openrazer and razergenie seem to work, but razergenie says "no devices detected" and I don't know why...

z3ntu, 2022-08-16:
Are you aware of this? https://github.com/openrazer/OBS-packaging/pull/8

szbinks, 2022-08-16:
Yes that's the first link I mentioned

z3ntu, 2022-10-15:
As I know practically nothing about Silverblue, I can't help unfortunately...

closed 2022-10-15 by z3ntu