openrazer/openrazer #1887: Support for [Razer Basilisk Essential]

tomasmark79, 2022-08-13:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

Etaash-mathamsetty, 2022-08-21:
This mouse is already supported, you must have misconfigured openrazer

z3ntu, 2022-10-15:
As mentioned in the previous comment, this device is already supported since v3.1.0. Check https://github.com/openrazer/openrazer/wiki/Troubleshooting for hints what might be going wrong.

closed 2022-10-15 by z3ntu