openrazer/openrazer #1889: Support for Razer DeathAdder V3 Pro

BLooD-DoG, 2022-08-20:
### Device Info

* Name: DeathAdder V3 PRO
* Product Number: RZ01-04630100
* Model Number: RZ01-0463

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
In Wired mode:

Bus 001 Device 007: ID 1532:00b6 Razer USA, Ltd Razer DeathAdder V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b6 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V3 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
[... 190 lines cut ...]
In Wired mode:
lrwxrwxrwx 1 root root  10 Aug 20 05:09 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Aug 20 05:09 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Aug 20 05:09 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Aug 20 05:09 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 Aug 20 05:09 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-mouse -> ../mouse2

In Wireless mode:

lrwxrwxrwx 1 root root  10 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if01-event-mouse -> ../event11
lrwxrwxrwx 1 root root   9 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if01-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if02-event-kbd -> ../event30
lrwxrwxrwx 1 root root   9 Aug 20 05:14 usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-mouse -> ../mouse2
```

### Packet Captures

Can provide if necessary

### Synapse Screenshots

Will provide in a reply. Suspect the driver will be very similar to the recent V2 Pro @ https://github.com/openrazer/openrazer/pull/1849/commits/a3fb25963fff4a0cd458abbaa4cd21a196accd6e since the mice are very similar.

BLooD-DoG, 2022-08-20:
I created a patch (attached) based on the V2 Pro commit, preliminary testing with the polychromatic frontend looks promising.

`diff --git a/README.md b/README.md
index 63ec6c7..f05cc9f 100644
--- a/README.md
+++ b/README.md
@@ -167,6 +167,8 @@ The devices below are fully feature supported by OpenRazer, which means all avai
 | Razer DeathAdder V2 X HyperSpeed              |  1532:009C  |
 | Razer Viper V2 Pro (Wired)                    |  1532:00A5  |
 | Razer Viper V2 Pro (Wireless)                 |  1532:00A6  |
+| Razer DeathAdder V3 Pro (Wired)               |  1532:00B6  |
+| Razer DeathAdder V3 Pro (Wireless)            |  1532:00B7  |
 
 ### Mousemats
 | Device                                        | USB VID:PID |
diff --git a/daemon/openrazer_daemon/hardware/mouse.py b/daemon/openrazer_daemon/hardware/mouse.py
index 7ae65a0..275ce8d 100644
--- a/daemon/openrazer_daemon/hardware/mouse.py
+++ b/daemon/openrazer_daemon/hardware/mouse.py
@@ -3039,3 +3039,49 @@ class RazerViperV2ProWireless(RazerViperV2ProWired):
     """
 
     USB_PID = 0x00A6
+
+class RazerDeathAdderV3ProWired(__RazerDevice):
+    """
+    Class for the Razer DeathAdder V3 Pro (Wired)
+    """
+    EVENT_FILE_REGEX = re.compile(r'.*usb-Razer_Razer_DeathAdder_V3_Pro_000000000000-if0(1|2)-event-kbd')
+
+    USB_VID = 0x1532
+    USB_PID = 0x00B6
+    METHODS = ['get_device_type_mouse', 'max_dpi', 'get_dpi_xy', 'set_dpi_xy', 'get_dpi_stages', 'set_dpi_stages',
+               'get_poll_rate', 'set_poll_rate',
+               'get_battery', 'is_charging', 'get_idle_time', 'set_idle_time']
+
+    DEVICE_IMAGE = "https://dl.razerzone.com/src/6130/6130-1-en-v2.png"
+
+    DPI_MAX = 30000
+
+    def __init__(self, *args, **kwargs):
+        super().__init__(*args, **kwargs)
+
+        self._battery_manager = _BatteryManager(self, self._device_number, 'Razer DeathAdder V3 Pro')
+        self._battery_manager.active = self.config.getboolean('Startup', 'battery_notifier', fallback=False)
+        self._battery_manager.frequency = self.config.getint('Startup', 'battery_notifier_freq', fallback=10 * 60)
+        self._battery_manager.percent = self.config.getint('Startup', 'battery_notifier_percent', fallback=33)
+
+    def _close(self):
+        """
+        Close the key manager
+        """
+        super()._close()
+
+        self._battery_manager.close()
+
+    def _resume_device(self):
+        self.logger.debug("Device doesn't have suspend/resume")
+
+    def _suspend_device(self):
+        self.logger.debug("Device doesn't have suspend/resume")
+
+
+class RazerDeathAdderV3ProWireless(RazerDeathAdderV3ProWired):
+    """
+    Class for the Razer DeathAdder V3 Pro (Wireless)
+    """
+
+    USB_PID = 0x00B7
diff --git a/driver/razermouse_driver.c b/driver/razermouse_driver.c
[... 194 lines cut ...]
index 7d688bc..4fa3b7a 100644
--- a/install_files/appstream/io.github.openrazer.openrazer.metainfo.xml
+++ b/install_files/appstream/io.github.openrazer.openrazer.metainfo.xml
@@ -90,6 +90,8 @@
     <modalias>usb:v1532p009Cd*</modalias>
     <modalias>usb:v1532p00A5d*</modalias>
     <modalias>usb:v1532p00A6d*</modalias>
+    <modalias>usb:v1532p00B6d*</modalias>
+    <modalias>usb:v1532p00B7d*</modalias>
     <modalias>usb:v1532p010Dd*</modalias>
     <modalias>usb:v1532p010Ed*</modalias>
     <modalias>usb:v1532p010Fd*</modalias>
diff --git a/install_files/udev/99-razer.rules b/install_files/udev/99-razer.rules
index 3a36926..0f9cf82 100644
--- a/install_files/udev/99-razer.rules
+++ b/install_files/udev/99-razer.rules
@@ -5,7 +5,7 @@ GOTO="razer_end"
 LABEL="razer_vendor"
 
 # Mice
-ATTRS{idProduct}=="0013|0016|0020|0024|0025|0029|002e|002f|0032|0034|0036|0037|0038|0039|003e|003f|0040|0041|0042|0043|0044|0045|0046|0048|004c|004f|0050|0053|0054|0059|005a|005b|005c|005e|0060|0062|0064|0065|0067|006a|006b|006c|006e|006f|0070|0071|0072|0073|0077|0078|007a|007b|007c|007d|0080|0083|0084|0085|0086|0088|008a|008c|008d|008f|0090|0091|0094|0095|0098|0099|009c|00a5|00a6", \
+ATTRS{idProduct}=="0013|0016|0020|0024|0025|0029|002e|002f|0032|0034|0036|0037|0038|0039|003e|003f|0040|0041|0042|0043|0044|0045|0046|0048|004c|004f|0050|0053|0054|0059|005a|005b|005c|005e|0060|0062|0064|0065|0067|006a|006b|006c|006e|006f|0070|0071|0072|0073|0077|0078|007a|007b|007c|007d|0080|0083|0084|0085|0086|0088|008a|008c|008d|008f|0090|0091|0094|0095|0098|0099|009c|00a5|00a6|00b6|00b7", \
     ATTRS{idVendor}=="1532", \
     ENV{ID_RAZER_CHROMA}="1", ENV{RAZER_DRIVER}="razermouse"
 `

Etaash-mathamsetty, 2022-08-21:
> I created a patch (attached) based on the V2 Pro commit, preliminary testing (wired and wireless) with the polychromatic frontend looks promising.
> 
> I wasn't sure where to get the proper DEVICE_IMAGE to use in mouse.py, best I could find was DEVICE_IMAGE = "https://dl.razerzone.com/src/6130/6130-1-en-v2.png" which seems to work ok in the front end.
> 
> [deathadderv3pro.patch.tar.gz](https://github.com/openrazer/openrazer/files/9388031/deathadderv3pro.patch.tar.gz)
> 
> [image]

the image that you have currently seems fine, since the mouse comes in two colors
and could you make a PR for this, if it works well?

closed 2022-11-09 by z3ntu