openrazer/openrazer #1892: Use single wireless dongle for multiple devices

J5cott, 2022-08-24:
### Distribution

Pop OS 22.04 LTS

Razer deathadder V2 pro
Razer BlackWidow V3 pro

### Suport

After boot, there is no way for a casual user to pair an existing dongle to a new device.  In order to make multiple wireless razer devices to work with your linux system you need to have each device's dongle plugged into the computer.  On windows you can plug  in one dongle and you are good to go.  Linux should have this same functionality.

Let me know what financial support is required to implement this feature.

z3ntu, 2022-10-15:
Sounds like #1869 and #1921 , right? Or are you talking about something different?

stlhrt, 2022-12-29:
Hyperpolling dongle is something different. 
Since some time the dongles shipped with some wireless devices allow pairing with multiple devices. Something akin to the logitech unified receiver.
Example hardware Deathstalker V2 Pro, Naga V2 Pro.
The dongle, when connected, report on `lsusb` as single device with ID of the device they came from, yet they are connected to up to two devices. 
Let me know if you need some debug info on this as I  have above 2 devices which can use each others dongles.

stlhrt, 2023-01-10:
pcap I think would help how it pairs or un pairs (like the standalone hyperspeed dongle)
The issue here is that on USB device id associated with a mouse or a keyboard, can exist another HID device which is another mouse or keyboard...  So if module to be loaded is detected by the identifier then second device in most cases will be a different kind, as most people would pair a keyboard to mouse dongle or vice versa.

stlhrt, 2023-01-11:
[nagav2pro_pairs_deathstalkerv2.pcapng.gz](https://github.com/openrazer/openrazer/files/10390352/nagav2pro_pairs_deathstalkerv2.pcapng.gz)

stlhrt, 2023-01-11:
[nagav2pro_UNpairs_deathstalkerv2.pcapng.gz](https://github.com/openrazer/openrazer/files/10390520/nagav2pro_UNpairs_deathstalkerv2.pcapng.gz)

z3ntu, 2023-01-11:
Indeed looks like it's sending requests to the different underlying devices
```
Status Trans Packet Proto DataSize Class CMD Args : nagav2pro_pairs_deathstalkerv2.pcapng
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    03       00    c5  0100a8 | UNKNOWN
02     1f    0000   00    03       00    bc  000000 | UNKNOWN
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    03       00    c5  0100a8 | UNKNOWN
02     1f    0000   00    03       00    bc  000000 | UNKNOWN
00     9f    0000   00    01       00    46  01 | UNKNOWN
00     9f    0000   00    03       00    41  010296 | UNKNOWN
00     9f    0000   00    01       00    0b  05 | UNKNOWN
02     1f    0000   00    03       00    c5  0100a8 | UNKNOWN
02     1f    0000   00    03       00    bc  000000 | UNKNOWN
02     1f    0000   00    31       00    bf  020100a80102960000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    03       00    c5  0100a8 | UNKNOWN
02     1f    0000   00    03       00    bc  000000 | UNKNOWN
```
Could you maybe do a screencap of doing these actions so I kind of understand how Razer shows this in their UI? And from what I understand now, you have the dongle for Razer Naga V2 Pro and then you somehow pair the Razer DeathStalker V2 Pro to the same wireless dongle and you can use both via this dongle? And Synapse also just pretends both devices are connected normally and nothing funky is going on?

As you can see in the part of the pcap above, there's quite some references to the PID of the Naga (`00a8`) and also some for the DeathStalker (`0296`) but honestly based on this limited sample size I can't really say what the rest of the parameters are supposed to mean or how I can figure out that one device is transaction id `0x1f` and the other `0x9f` apparently..

stlhrt, 2023-01-12:
https://user-images.githubusercontent.com/3652939/211944567-26636cbd-0133-42f1-9bfe-b4c5916d8621.mp4
[nagav2pro_pairs_deathstalkerv2-more.pcapng.gz](https://github.com/openrazer/openrazer/files/10396914/nagav2pro_pairs_deathstalkerv2-more.pcapng.gz)

The process is very simple both on the Naga dongle and DS one.  Just click `Pair` turn on device and it connects.

z3ntu, 2023-01-15:
Clear, thanks!

A few more questions: if the two devices are paired and you turn them off, does Synapse indicate that it's turned off? If so can you make a pcap (with both devices connected), then turn off the mouse and maybe wait a bit. And a separate one where you turn off the keyboard?

Also do you have another device other than your keyboard that you can pair with this dongle? Would be good to see what arguments change in the pcap as many parameters are still unclear.

stlhrt, 2023-01-15:
Just the mouse and keyboard. I can pair either device to the others dongle
and that's it.
As for synapse, when two devices are sharing one dongle either one can be
turned off and it disappears from there application.
I have them both wireless on single dongle all the time and use cable to
charge them when needed.
When connecting with cable the device blinks in synapse as if connected as
different device :-)

z3ntu, 2023-01-15:
Pcap would be nice then for the case where one device disappears when turned off :) Just to confirm my suspicions I see in the args

stlhrt, 2023-01-16:
[dsv2pro_and_naga_synapse_on_off_interactions.pcapng.gz](https://github.com/openrazer/openrazer/files/10421675/dsv2pro_and_naga_synapse_on_off_interactions.pcapng.gz)

pcap shows how synapse detects and interrogates dongle with 2 devices attached. Then how it behaves when either one is turned on and of in different order. Hope this helps :)

z3ntu, 2023-01-16:
```
02     88    0000   00    31       00    bf  020100a80102960000000000000000000000000000000000000...
02     1f    0000   00    31       00    bf  020100a80102960000000000000000000000000000000000000...
02     1f    0000   00    31       00    bf  020100a80102960000000000000000000000000000000000000...
02     1f    0000   00    31       00    bf  020100a80002960000000000000000000000000000000000000...
```
So based on these packets I would assume both devices are connected in the first three, and then in the last the `0296` device (keyboard) disappears. But based on your description you also turned off the mouse in the meantime so it doesn't seem to be exactly that way...

stlhrt, 2023-01-17:
so you'd need more precise sample with exact descriptions. Maybe a pcap with a screen capture showing whats going in the synapse UI?

stlhrt, 2023-01-18:
[dsv2pro_and_nagav2pro-2023-01-19_00.21.43.pcapng.gz](https://github.com/openrazer/openrazer/files/10451993/dsv2pro_and_nagav2pro-2023-01-19_00.21.43.pcapng.gz)

https://user-images.githubusercontent.com/3652939/213318293-5ec076ca-f913-4bd4-b88c-d422039f240e.mp4

This pair should be more helpful :)

z3ntu, 2023-02-11:
fwiw these are the 00-bf packets from the new capture
```
Status Trans Packet Proto DataSize Class CMD Args
02     08    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    31       00    bf  020100a800ffff0000000000000000000000000000000000000... | UNKNOWN
02     1f    0000   00    31       00    bf  020100a80102960000000000000000000000000000000000000... | UNKNOWN
02     08    0000   00    31       00    bf  020000a80102960000000000000000000000000000000000000... | UNKNOWN
```

really seems to be a not-so-trivial feature that I don't really have the time to look into in-depth...

J5cott, 2023-03-03:
Should I close this issue since it is not going to be implemented?

johnafogarty4, 2023-03-09:
damn, i was hopeful that this would be something that could be implemented.

z3ntu, 2023-03-12:
@J5cott How do you get this impression? I doubt I implied anywhere that this is out-of-scope for OpenRazer

J5cott, 2023-03-12:
I guess I misread your comment: "...really seems to be a not-so-trivial feature that I don't really have the time to look into in-depth" lead me to believe that this wasn't going to happen

z3ntu, 2023-03-12:
Pull requests are always welcome :) Generally this project nowadays is mostly driven by pull requests, at least for devices and new device-specific features since for the most part you need to have the access to the device to develop and test.

jarrard, 2023-06-22:
I have a Razer BlackWidow3 Pro connected to my Razer Orochi V2 receiver and it seems Polychromatic and RazerGenie will only show the mouse device and not keyboard thus I can't adjust the RGB on the BlackWidow. 

It's a bit of a shame, seems related to this. If there is a workaround  let me know. 

Cheers