openrazer/openrazer #1893: Device not detected.

mlanca, 2022-08-25:
A 1532:0A24 is not detectable.  Tried poly & genie.  meta package installed.  using Manjaro.  Works fine under Win10(amd 4600g w.12g)

lah7, 2022-08-25:
[What troubleshooting have you tried?](https://github.com/openrazer/openrazer/wiki/Troubleshooting) In Polychromatic, click **Troubleshoot** to find out which component is not installed properly.

mlanca, 2022-12-24:
hi - I had a pretty massive stroke.  However, if you can spare the time. P
would love ro get this working.  Let me know how you would like to
proceed.  It works with Windows.  However, it doesn't see the keyboard in
Linux.

Sincerely,
Mike Lancaster

z3ntu, 2022-12-28:
If you could try following what @lah7 wrote that would be helpful. Or check this wiki page https://github.com/openrazer/openrazer/wiki/Troubleshooting

closed 2023-05-28 by github-actions[bot]