openrazer/openrazer #1894: Support for Razer Kraken V3

LukeSerne, 2022-08-25:
### Device Info

* Name: Razer Kraken V3
* Product Number: `0549`
* Model Number: <See under device>

### Device Information
```
$ lsusb -d 1532: -v
Bus 001 Device 007: ID 1532:0549 Razer USA, Ltd Razer Kraken V3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0549 
  bcdDevice            0.04
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V3
  iSerial                 3 000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00f7
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004f
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
[... 206 lines cut ...]
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```

```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 120 25 aug 13:29 .
drwxr-xr-x 4 root root 560 25 aug 13:29 ..
lrwxrwxrwx 1 root root  10 25 aug 13:29 usb-Razer_Razer_Kraken_V3_000000000001-event-if03 -> ../event17
```

### Packet Captures / Synapse Screenshots

I don't have a windows environment to collect pcaps / screenshots from Razer Synapse.

z3ntu, 2022-10-15:
Hi, does this device have any model number and product number like e.g. RZ04-0375? Might be printed on the packaging or somewhere else on the device.

LukeSerne, 2022-10-15:
The model number appears to be `RZ04-0377`. This was printed on the packaging. I have updated my initial message to include the model number.

Dekomoro, 2023-04-18:
I have a windows environment to capture pcaps and synapse screenshots, I'd need a guide on doing so, but I'd be up to contributing to get my headset working better on linux

z3ntu, 2023-04-22:
@Dekomoro The first parts of https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol cover that.

TallGirlVanessa, 2023-06-21:
<details>
<summary>
Synapse screenshots
</summary>

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

</details>

TallGirlVanessa, 2023-06-22:
For pcap, I have a Windows 11 machine, can I use USBPcap, or do I need to do the VM dance? I think I'd need to upgrade to Pro to generate Windows VMs, now that the IE VMs are no longer provided by Microsoft.

z3ntu, 2023-07-16:
@TallGirlVanessa Late answer, but wireshark should also work directly, but I've never tried it myself.

But you should be able to easily get a Windows 10 or 11 VM from Microsoft directly.

PTony, 2023-07-31:
Hi there,

I've captured some packets for the Kraken V3. I hope i've done it well. :)

[Kraken_v3_pcaps.zip](https://github.com/openrazer/openrazer/files/12216066/Kraken_v3_pcaps.zip)

z3ntu, 2023-08-06:
@PTony Thanks but my script doesn't appear to find any packets that matches the known packets. 

Or alternatively you can look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md and implement support for this device directly, should with trial & error also for some parameters, but probably guessing based on already supported devices is enough.

PTony, 2023-08-14:
Thx @z3ntu . I take note of that. 

Eventually I will have a look at it in few weeks, but it's not a promise at all. 

If someone else want to give it a go don't wait after me.

Homelessguy42, 2024-05-26:
Hello! I created a fork of this project that attempts to add support for the Razer Kraken V3 headset. Here is the link to the fork: 

https://github.com/Homelessguy42/openrazer

 My project was able to be be built under a Debian virtual machine without any issues and here is my relevant dmesg output `[  991.103508] razeraccessory: loading out-of-tree module taints kernel.
[  991.103555] razeraccessory: module verification failed: signature and/or required key missing - tainting kernel
` following that, here is the output of `PYTHONPATH="pylib:daemon" python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf`  

`user1@workstation:~/openrazer-branch/openrazer$ PYTHONPATH="pylib:daemon" python3 ./daemon/run_openrazer_daemon.py -Fv --config=$PWD/daemon/resources/razer.conf
Starting daemon.
2024-05-26 19:37:23 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 8410
2024-05-26 19:37:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWired ----------------- (1532:00AA)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWireless -------------- (1532:00AB)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3XHyperSpeed -------------- (1532:00B9)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2024-05-26 19:37:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
[... cut ...]
2024-05-26 19:37:23 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2024-05-26 19:37:23 | razer                          | INFO     | Serving DBus
`

Now here are my questions: How can I test this further? (e.g. controlling the LEDs on the headset)

and how can I add the associated image of the headset? If anyone has any questions / needs more information please let me know! I want to do everything I can to get this headset supported under GNU/Linux

z3ntu, 2024-06-05:
@Homelessguy42 The simplest way is probably to install a UI like polychromatic or RazerGenie and click around and see if the different effects and everything works.

lyricalpaws, 2024-10-05:
still waiting for this tbh, having my headset disconnect and reconnect at a whisper of a volume constantly any time I boot something is very annoying

z3ntu, 2024-10-06:
@lyricalpaws OpenRazer support wouldn't fix any of this, we don't touch the audio stack at all, just add some functionality on top like normally lights and some other misc. settings.

paladin-devops, 2025-01-06:
@homelessguy42 any updates on your fork? I'm working on supporting the Kraken V3 Pro and running into issues, seeking inspiration for a fix. (#1950)

OnlyZaenae, 2025-01-12:
Any updates on the progress? Or is it abandoned? I would help (as a sofware engineer), but I worked in this area never.

Homelessguy42, 2025-01-12:
> @Homelessguy42 any updates on your fork? I'm working on supporting the Kraken V3 Pro and running into issues, seeking inspiration for a fix. (#1950)

Nothing of substance in terms of an update. I attempted to add support myself but I'm still little unsure how to go about testing it. I would've spent more time on it but I've had very pressing family events to handle so my workflow was completely disrupted. If you have knowledge that I don't, feel free to take a crack at it. The fork is on my GitHub.

hendy643, 2025-01-26:
@Homelessguy42 I'd like to help, if I can. I've a little (read: tiny) experience as I did the support for the Naga Left Handed Edition. But this seems like a bigger thing to sort out.
I'll try and capture some wireshark data in the next few days.

Homelessguy42, 2025-01-28:
> [@Homelessguy42](https://github.com/Homelessguy42) I'd like to help, if I can. I've a little (read: tiny) experience as I did the support for the Naga Left Handed Edition. But this seems like a bigger thing to sort out. I'll try and capture some wireshark data in the next few days.

If I understand correctly, someone already uploaded their packet captures but how useful those ones are, is a bit unclear to me. Wishing you luck!

cruzisonfire, 2025-03-03:
Thank you all for figuring this out- as someone who cares but not as much its much appreciated!

RjGunner111, 2025-06-23:
i got the pcap right here

[razer kraken v3.tar.gz](https://github.com/user-attachments/files/20873471/razer.kraken.v3.tar.gz)

RjGunner111, 2025-06-24:
well, the file extension is pcapng, but if the extension must be a pcap, i'll see what i can do.

RjGunner111, 2025-06-27:
i have managed to get the program to recognize the device. however, whenever i try to change the effects, i can't for the life of me figure out why the effects won't change. i've looked inside the driver code, daemon code, everything seems to be in order.

[image]

RjGunner111, 2025-06-27:
i discovered while i was testing the device in terms of changing the color, it turns out fsr, the firmware and serial number shows as stuff that i doubt is right... the firmware claims it's v9.99

RjGunner111, 2025-06-28:
i have uploaded my fork for this, lemme know if this works better for anyone:

https://github.com/RjGunner111/openrazer