openrazer/openrazer #1895: "$ openrazer-deamon -Fv" displays "OSError: [Errno 22] Invalid Argument" resulting in unregocnized device

cemno, 2022-08-25:
### Distribution

Ubuntu 20.04; Kernel 5.18.19-xanmod1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

### Issue

So i noticed my device is not being recognized (its been like that for quite a while now, but i never got to troubleshoot the issue) and opened polychromatic to check on the device. It says 1532:0016 and unrecognized Device. I restarted the OpenRazer Backend which did not help. Using the Troubleshot button build into Polychromatic. It displays all tests passed but "Deamon is running". I entered the command suggested there to troubleshoot the problem (`$ openrazer-deamon -Fv`). Some errors appeared which stopped the deamon, but i don't understand what they mean. :(

Here is the output: 
[error_log_openrazer_daemon.log](https://github.com/openrazer/openrazer/files/9424247/error_log_openrazer_daemon.log)

Afterwards i followed the Troubleshooting wiki, which (Step 1) marks my (quite old) device as supported and that (Step 2) the kernel module installed is installed. 

Output of command in Step 2:
`Module openrazer-driver/3.4.0 already installed on kernel 5.18.19-xanmod1/x86_64`

Step 3 didn't worked at the beginning, but reinstalling openrazer and restarting the PC a few times modprobing seems to work now, so `sudo modprobe razerkdb` does not have an output.

As mentioned in Step 4 plugdev ist given permission and writes the log as expected.

I can execute all the log commands in Step 5, but they all seem to give at least some reasonable output, but i obviously don't know if it is correct. 

So following all the steps i should have a device that should be recognized and working, but some `OSError: [Errno 22] Invalid Argument` displayed by the `$ openrazer-deamon -Fv` command seems to deny that, see attached file mentioned earlier.

I hope somebody can help me here, as i'm not sure whats the problem with my system so that i'm able to change the DPI setting again. :cry:

lah7, 2022-08-25:
Your installation appears to be okay. There's just a problem sending the DPI command via the driver file, and then daemon crashes as a result.

This is the exception of interest:
```python
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 117, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 491, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 247, in __init__
    dpi_func(self.dpi[0], self.dpi[1])
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/mamba.py", line 197, in set_dpi_xy
    driver_file.write(dpi_bytes)
OSError: [Errno 22] Invalid argument
```

It could be as a result of changes since the previous version, try:

* Deleting the persistence file: `~/.config/openrazer/persistence.conf` (hidden folder in home directory)
* Or, disable the persistence feature in `~/.config/openrazer/razer.conf`:
    ```diff
    -restore_persistence = True
    +restore_persistence = False
    ```

Then see if you can run `openrazer-daemon -F`  again.

cemno, 2022-08-26:
Thank you, that fixed it! :smiley: 

Upon seeing the persistence.conf file i think i remember that i created or adjusted it because there was a bug back than not being able to change DPI with the Deathadder 3.5G. And 3500DPI was definitely way to much... :sweat_smile: 

But everything is working now! Thank you very much once again for the nice support.

closed 2022-08-26 by cemno