openrazer/openrazer #1897: razermouse kernel module causes random keyboard input failures

pdf, 2022-08-26:
### Distribution

Arch Linux
openrazer-driver-dkms 3.4.0-1
Xorg

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Having recent versions of the razermouse driver loaded appears to periodically break keyboard input (for a totally unrelated input device). The symptom is that key inputs are randomly dropped, and not loading the razermouse kmod causes the problem to stop.

lah7, 2022-08-28:
Which kernel version are you using? Anything in the logs?

I have two partitions of Arch Linux on my machine - one for work (`linux 5.19.4`) and one for home (`linux-lts 5.15.63`). I thought something was a little off the past week with my <kbd>e</kbd> key occasionally missing a keypress on the work partition, but no issues with the home partition. Only difference here are the kernel versions. 

If I notice a keypress miss next week, I'll try unloading `razermouse`. I assumed dust in the key switch was the culprit. Both my keyboard and mouse are Razer.

pdf, 2022-08-28:
I think the last kernel version I explicitly tested it on was 5.18.16, but I've just gotten around to reporting it.

I'm currently on drm-tip for some Intel issues, but I can try enabling it here again to try to reproduce and grab logs. It can take some time before the problem occurs though, so might be a while before I can tell whether it's still evident or not.

The affected devices:
1038:1122 SteelSeries ApS SteelSeries KLC
1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]

lah7, 2022-09-07:
I uninstalled OpenRazer on my work partition (`linux-5.19.6`) for a while this week, and still seem to get the odd mispresses. 

Makes me wonder if there's something in the latest kernel, drivers or general stack that's just affecting input in general. NVIDIA, X.org, KDE desktop here.

pdf, 2022-09-07:
It's happened here once a couple of days ago while running openrazer on drm-tip, but I didn't see anything useful in the logs. I've removed this driver now, and waiting to see if it happens again. but considering that repro took a week, could be a while.

It's entirely possible the bug is somewhere else, I just wonder if with my hardware the openrazer driver makes it easier to trigger, because I haven't seen it happen yet without the driver.

We'll see over the next couple of weeks if I can repro here again without it, then I'll try again with the driver loaded, it's not easy to debug though when it happens in the middle of work, and half your keypresses don't land.

pdf, 2022-10-22:
This turned out to be a hardware problem, apologies for the noise.

closed 2022-10-22 by pdf