openrazer/openrazer #1898: Arch derivative fails to validate module

LividJava, 2022-08-27:
### Distribution
EndeavourOS  5.19.3-arch1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Polychromatic reports that the Kernel modules are not installed.
[image]

However I do have failed to validate module from dmesg
[image]

Also polychromatic reports an old version of openrazer installed when yay reports the latest version is installed.

Uncertain if this is related to #1893 but it only started happening after a system update.

lah7, 2022-08-28:
I found this [Arch bug report](https://bugs.archlinux.org/task/75685) showing a similar error affecting kernel 5.19.3. The solution there was to downgrade `pahole` and then cleanly reinstall the module (`openrazer-driver-dkms` in our case). Does that work?

LividJava, 2022-08-28:
Downgrading `pahole` to `1:1.23-1` did work can confirm

closed 2022-08-28 by LividJava