openrazer/openrazer #1900: Basilisk V3 not detected by UEFI firmware or system after reboot

jfroy, 2022-08-28:
### Distribution and hardware

Distro: Arch Linux
Linux: 5.19.4-arch1-1
OpenRazer: 3.4.0
Motherboard: ROG CROSSHAIR VIII DARK HERO
MB firmware/BIOS: 4201 (latest)
Boot: UEFI secure boot (shim + signed unified kernel image)
Mouse: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After configuring my mouse with OpenRazer (via Polychromatic), it not longer works after rebooting, not even in the UEFI interface. Unplugging and re-plugging the device does make it work until the next reboot.

Looking at the kernel logs, you can see some really basic descriptor read errors, which makes me suspect the firmware is somehow misconfigured.

```
[    2.725429] usb 3-2.2: new full-speed USB device number 4 using xhci_hcd
[    2.815748] usb 3-2.2: device descriptor read/64, error -32
[    3.015745] usb 3-2.2: device descriptor read/64, error -32
[    3.215706] usb 3-2.2: new full-speed USB device number 5 using xhci_hcd
[    3.309068] usb 3-2.2: device descriptor read/64, error -32
[    3.509063] usb 3-2.2: device descriptor read/64, error -32
[    3.616247] usb 3-2-port2: attempt power cycle
[    4.229034] usb 3-2.2: new full-speed USB device number 6 using xhci_hcd
[    4.229521] usb 3-2.2: Device not responding to setup address.
[    4.435806] usb 3-2.2: Device not responding to setup address.
[    4.642361] usb 3-2.2: device not accepting address 6, error -71
[    4.735714] usb 3-2.2: new full-speed USB device number 7 using xhci_hcd
[    4.735822] usb 3-2.2: Device not responding to setup address.
[    4.942464] usb 3-2.2: Device not responding to setup address.
[    5.149034] usb 3-2.2: device not accepting address 7, error -71
[    5.149610] usb 3-2-port2: unable to enumerate USB device
```

Full device USB info:

```Bus 003 Device 014: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0099 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk V3
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

jfroy, 2022-08-28:
Device settings (from `persistence.conf`):

```[632139H21310224]
dpi_x = 1800
dpi_y = 1800
poll_rate = 1000
backlight_active = True
backlight_brightness = 50
backlight_effect = spectrum
backlight_colors = 0 255 0 0 255 255 0 0 255
backlight_speed = 1
backlight_wave_dir = 1
logo_active = True
logo_brightness = 50
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1
scroll_active = True
scroll_brightness = 50
scroll_effect = spectrum
scroll_colors = 0 255 0 0 255 255 0 0 255
scroll_speed = 1
scroll_wave_dir = 1

```

z3ntu, 2022-10-15:
All this low-level USB stuff is completely out of control of OpenRazer. But of course you can try uninstalling openrazer to see if it changes anything, if not it's not caused by openrazer.

closed 2023-03-18 by github-actions[bot]