openrazer/openrazer #1902: Support for Razer Seiren V2 X

nikzad-avasam, 2022-08-30:
### Device Info

* Name: Razer Seiren V2 X
* Product Number: 0543
* Model Number: V2 X

### Device Information
```
Bus 001 Device 004: ID 1532:0543 Razer USA, Ltd Razer Seiren V2 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        16
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0543 
  bcdDevice           10.15
  iManufacturer           1 Razer Inc
  iProduct                2 Razer Seiren V2 X
  iSerial                 3 UC2138L04403263
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0131
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass          1 Audio
      bFunctionSubClass       0 
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol     32 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        bCategory               4
        wTotalLength       0x0089
        bmControls           0x00
      AudioControl Interface Descriptor:
        bLength                17
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bCSourceID             16
[... 270 lines cut ...]
```

```
~ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 160 Aug 31  2022 .
drwxr-xr-x 4 root root 560 Aug 31  2022 ..
lrwxrwxrwx 1 root root   9 Aug 31  2022 usb-Logitech_USB_Optical_Mouse-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Aug 31  2022 usb-Logitech_USB_Optical_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Aug 31  2022 usb-Razer_Inc_Razer_Seiren_V2_X_UC2138L04403263-event-if03 -> ../event9
lrwxrwxrwx 1 root root   9 Aug 31  2022 usb-SINO_WEALTH_Gaming_KB-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Aug 31  2022 usb-SINO_WEALTH_Gaming_KB-event-kbd -> ../event4
lrwxrwxrwx 1 root root  10 Aug 31  2022 usb-Sonix_Technology_Co.__Ltd._USB_2.0_Camera-event-if00 -> ../event21
~ 

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
this is in windows and work fine . There is an option for microphones that is enabled by default in Linux and it has caused the quality of the microphone to drop : 
[image]

z3ntu, 2022-10-15:
I kind of doubt these settings are handled by the standard Razer protocol for lighting and such, but if you can give me USB packet captures from Windows (see wiki page) then I can double check.

daniel-bogdoll, 2023-01-26:
I would also be interested in the V2 Pro (https://www.razer.com/streaming-microphones/razer-seiren-v2-pro)

z3ntu, 2023-02-11:
@daniel-bogdoll If you have that device and it has a different USB PID (not 1532:0543) please create a new issue for that device if it doesn't exist already.

fmuyassarov, 2023-05-20:
I just got my new Razer Seiren V2 without even thinking that that would be any issues with drivers on Ubuntu but I run into the exact same issue which is quite disappointing 

[image]
[image]

fmuyassarov, 2023-05-20:
@z3ntu is there any plan to have the support for Razer Seiren V2 in the near future?

z3ntu, 2023-05-30:
@fmuyassarov If somebody adds support for it via a pull request then it'll be supported. But as written above in https://github.com/openrazer/openrazer/issues/1902#issuecomment-1279783665 I don't think this device needs any kind of OpenRazer support, in what it can support.