openrazer/openrazer #1903: Device 1532:0096 not supported

iamagarre, 2022-08-30:
Hi,

this is my first support ticket so I hope to do it right.

I currently have a razer keyboard and a razer mouse. Mouse is a Razer Naga X, the keyboard is a Razer Blackwidow v3. The keyboard works fine with openrazer and polychromatic but the mouse shows a message that is not supported. I have readed in another support issue that support for Razer Naga X was already added but mine is not working.

Any idea why keyboard works and mouse does not?

Best Regards

Etaash-mathamsetty, 2022-08-31:
you have to compile openrazer from this PR https://github.com/openrazer/openrazer/pull/1878, while we wait for it to get merged and become part of the next release

iamagarre, 2022-08-31:
Thank you and sorry for the duplicate.

z3ntu, 2022-10-15:
Closing then.

closed 2022-10-15 by z3ntu