openrazer/openrazer #1904: Are there 2 or 3 modes?

flarn2006, 2022-09-08:
https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver

> There are 3 available modes:
> 
> * 0x00, 0x00: Normal Mode
> * 0x03, 0x00: Driver Mode

Where's the third one?

z3ntu, 2022-10-15:
There's also a bootloader/dfu mode but since it's easy to brick devices with that it's not documented. Any other questions?

flarn2006, 2022-10-15:
Isn't that more of a reason *to* document it? So people know how best to
avoid that risk if they still need to use that mode for something.

z3ntu, 2022-10-15:
I've never tried putting my devices into this mode so I honestly cannot say how it behaves. I just know from the early days that there have been reports of bricked devices, I don't know any more specifics about this though.

closed 2023-03-16 by github-actions[bot]