openrazer/openrazer #1905: Create a snap package?

heidiwenger, 2022-09-09:
Many can't install OpenRazer as of now, it is too difficult for a normal user. PPA's are a thing of the past. Maybe it is time to attempt a creation of a simple package with all dependencies and in secure sandbox? 

Why not aim for a simple user friendly way to install on Linux? A snap package also enables an easy roll back or a version in case of need. Ubuntu is the main Linux for new users, and the old way of installing would still remain of course!

Every single part of Snap that you come into contact with is open source. The package format, the daemon, the website (snapcraft.io), and the Snap Store that runs on your system. The only part that is not is some piece of the backend on Ubuntu's servers.

Could a snap package by any chance be possible then? A simple one click install, that is. Thanks.

Etaash-mathamsetty, 2022-09-09:
Not many people like snaps or even use snaps, and it's something extra that we have to install for no real reason (except on ubuntu), creating a flatpak is more advisable and 99% of linux distros pick flatpak over snap. 
The other problem is that openrazer provides a kernel module, making it impossible to ship with as a snap, flatpak or AppImage. 
It would be possible to get polychromatic to make a flatpak though.

z3ntu, 2022-09-09:
As in the previous comment

> And then the next problem is that openrazer provides a kernel module, making it impossible to ship with as a snap, flatpak or AppImage.

See also #742

closed 2022-09-09 by z3ntu

heidiwenger, 2022-09-09:
> Not many people like snaps or even use snaps, and it's something extra that most linux users would have to install for no real reason (except on ubuntu), creating a flatpak is more advisable and 99.9% of linux distros pick flatpak over snap. The other problem is that flatpak is more popular, [Insert all other distros except ubuntu here] > ubuntu in terms of total users. And then the next problem is that openrazer provides a kernel module, making it impossible to ship with as a snap, flatpak or AppImage. It should be possible to get polychromatic to make a flatpak though. PS: If you are finding using PPA's difficult, anything arch based will probably satisfy your needs thanks to the AUR

Snaps are actually far more popular than flatpaks. Look it up. There are also 1500 flatpaks and nearly 7000 snap packages now. Developers especially embrace snaps, despite the common misconception. 

That module kernel thing is only a flatpak problem: you can ship anything from libraries, server/CLI softare to a Linux kernel itself as a snap. See, flatpak and snapd are not very common. Snaps are more versatile. Flatpaks are only for desktop you see. Snaps also use less RAM (another misconception) than flatpak. 

Ubuntu is the most used distro, and a normal user wants a simple install. They don't care if Firefox starts currently 2 seconds slower than the deb package. Snaps are always in sandbox, so more secure for a new user.

Karagra, 2022-09-09:
> Snaps are actually far more popular than flatpaks. Look it up. There are also 1500 flatpaks and nearly 7000 snap packages now. Developers especially embrace snaps, despite the common misconception. [...]

Snap is not as popular as you think.

Etaash-mathamsetty, 2022-09-09:
> Snaps are actually far more popular than flatpaks. Look it up. There are also 1500 flatpaks and nearly 7000 snap packages now. Developers especially embrace snaps, despite the common misconception. [...]

1. Developers do not embrace snaps (the number of snaps literally doesn't prove anything, for example if I have 50 1 m^3 cubes vs 25 50 m^3 cubes, the latter has more volume), only the developers that are paid by canonical to use snaps do, EVERYBODY is migratiing over to flatpak now. 
Kernel modules cannot be shipped with snap, that is simply impossible, in fact it will never be possible to ship kernel modules that work across distros in a solution like flatpak or snap or app image.

DefaultX-od, 2022-09-10:
> > > Not many people like snaps or even use snaps, and it's something extra that most linux users would have to install for no real reason (except on ubuntu), creating a flatpak is more advisable and 99.9% of linux distros pick flatpak over snap. The other problem is that flatpak is more popular, [Insert all other distros except ubuntu here] > ubuntu in terms of total users. And then the next problem is that openrazer provides a kernel module, making it impossible to ship with as a snap, flatpak or AppImage. It should be possible to get polychromatic to make a flatpak though. PS: If you are finding using PPA's difficult, anything arch based will probably satisfy your needs thanks to the AUR
> > 
> > 
> > Snaps are actually far more popular than flatpaks. Look it up. There are also 1500 flatpaks and nearly 7000 snap packages now. Developers especially embrace snaps, despite the common misconception.
> > That module kernel thing is only a flatpak problem: you can ship anything from libraries, server/CLI softare to a Linux kernel itself as a snap. See, flatpak and snapd are not very common. Snaps are more versatile. Flatpaks are only for desktop you see. Snaps also use less RAM (another misconception) than flatpak.
> > Ubuntu is the most used distro, and a normal user wants a simple install. They don't care if Firefox starts currently 2 seconds slower than the deb package. Snaps are always in sandbox, so more secure for a new user.
> 
>     1. Developers do not embrace snaps (the number of snaps literally doesn't prove anything, for example if I have 50 1 m^3 cubes vs 25 50 m^3 cubes, the latter has more volume), only the developers that are paid by canonical to use snaps do, EVERYBODY is migrating over to flatpak now.
> 
>     2. Kernel modules cannot be shipped with snap, that is simply impossible, in fact it will never be possible to ship kernel modules that work across distros in a solution like flatpak, snap or appimage.
> 
>     3. popularity of one distro compared to another doesn't matter here, it's a battle between 1 distro and 1000s of other distros that choose flatpak. If we assume Ubuntu is the only distro that ships snaps (which it is), then that's only 13% of distros with snaps and the other 87% with flatpak.
> 
>     4. flatpak has sandboxing as well, and it probably works better since it doesn't use apparmor (which won't work correctly on other distros without extra configuration)

1. Canonical must be really rich if they can afford to pay JetBrains, Microsoft, Google, Mozilla, Wildfire Games, Postman inc, Mailspring, Spotify, Slack, gitkraken, Penguin Labs, and various other publishers, to upload their apps to the Snap Store... Or maybe they do embrace snaps.
2. Maybe it's true, but you are clearly not a reliable source of that kind of information. Let's just wait, till Canonical starts to ship newer kernels as snaps, and let's see what happens to the kernel modules.
3. Well, if 1 distro accounts for 50~60% of the marketshare, you'll need to consider what is available in that distro in regard to the tools used to deliver apps to the users, of course if you want to reach as many people as possible.
4. -

heidiwenger, 2022-09-10:
> > > Not many people like snaps or even use snaps, and it's something extra that most linux users would have to install for no real reason (except on ubuntu), creating a flatpak is more advisable and 99.9% of linux distros pick flatpak over snap. The other problem is that flatpak is more popular, [Insert all other distros except ubuntu here] > ubuntu in terms of total users. And then the next problem is that openrazer provides a kernel module, making it impossible to ship with as a snap, flatpak or AppImage. It should be possible to get polychromatic to make a flatpak though. PS: If you are finding using PPA's difficult, anything arch based will probably satisfy your needs thanks to the AUR
> > 
> > 
> > Snaps are actually far more popular than flatpaks. Look it up. There are also 1500 flatpaks and nearly 7000 snap packages now. Developers especially embrace snaps, despite the common misconception.
> > That module kernel thing is only a flatpak problem: you can ship anything from libraries, server/CLI softare to a Linux kernel itself as a snap. See, flatpak and snapd are not very common. Snaps are more versatile. Flatpaks are only for desktop you see. Snaps also use less RAM (another misconception) than flatpak.
> > Ubuntu is the most used distro, and a normal user wants a simple install. They don't care if Firefox starts currently 2 seconds slower than the deb package. Snaps are always in sandbox, so more secure for a new user.
> 
> Snap is not as popular as you think, If they fix all the issues I could see the opinion of snap changing.. but as it stands flatpak just works

I just checked the numbers given by Alan Pope many times. It is not kinda of a opinion or "think hunch". They sure are popular, in contradiction of what picture some discussions might give. The devs love snaps and a normal user just want to install simple and secure. No matter if Firefox currently starts 2 seconds longer...

Etaash-mathamsetty, 2022-09-11:
> 1. Canonical must be really rich if they can afford to pay JetBrains, Microsoft, Google, Mozilla, Wildfire Games, Postman inc, Mailspring, Spotify, Slack, gitkraken, Penguin Labs, and various other [...]

1. Fun fact, they did in fact pay those developers, and most of those have an unofficial flatpak port
2. Well that's true, but just based on how these packaging formats work, it doesn't take that many braincells to figure out that kernel modules and operating system kernels can't be installed through a universal packaging format.
3. This 1 distro accounts for nearer to 20% rather than 60.

Etaash-mathamsetty, 2022-09-11:
> I just checked the numbers given by Alan Pope many times. It is not kinda of a opinion or "think hunch". They sure are popular, in contradiction of what picture some discussions might give. The devs [...]

I'm not saying snaps are not popular, I'm just saying they are less popular than something like flatpak.
Flatpak will give the same experience that snap will, it simple and secure just like snap. 
I'm not against snaps, it's just that it's a better idea to use something that's more open, popular, and not as controversial in the community

lah7, 2022-09-11:
On that note, let's lock this issue. OpenRazer's stance is, "contributions welcome" (#742). Be prepared that this might not be possible right now due to this project being a **kernel module**.

In future, please avoid these massive quote replies. This made the issue difficult to read, and clogs up the Telegram GitHub group. Thanks!