openrazer/openrazer #1912: Support for Razer Deathadder v2 lite

Mario156090, 2022-09-21:
### Device Info

* Name: Razer Deathadder v2 lite
* Product Number: RZ01-03340100
* Model Number: v2 lite

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
`[mario@Asus-Arch ~]$ lsusb -d 1532: -v

Bus 003 Device 002: ID 1532:00a1 Razer USA, Ltd Razer DeathAdder V2 Lite
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a1 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V2 Lite
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
[mario@Asus-Arch ~]$ 
`
```

```
Please post the output of
    ls -lah /dev/input/by-id/
`[mario@Asus-Arch ~]$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 set 20 22:35 .
drwxr-xr-x 4 root root 540 set 20 22:35 ..
lrwxrwxrwx 1 root root  10 set 20 22:35 usb-Logitech_Wireless_Gamepad_F710_55EA975C-event-joystick -> ../event17
lrwxrwxrwx 1 root root   6 set 20 22:35 usb-Logitech_Wireless_Gamepad_F710_55EA975C-joystick -> ../js0
lrwxrwxrwx 1 root root  10 set 20 22:35 usb-Razer_Razer_DeathAdder_V2_Lite-event-if01 -> ../event13
lrwxrwxrwx 1 root root   9 set 20 22:35 usb-Razer_Razer_DeathAdder_V2_Lite-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 set 20 22:35 usb-Razer_Razer_DeathAdder_V2_Lite-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 set 20 22:35 usb-Razer_Razer_DeathAdder_V2_Lite-if02-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 set 20 22:35 usb-Razer_Razer_DeathAdder_V2_Lite-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 set 20 22:35 usb-Sonix_Technology_Co.__Ltd._USB2.0_HD_UVC_WebCam-event-if00 -> ../event12
`

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
pending

z3ntu, 2022-10-15:
This might just be a south-american variant of #1258 from what I could find, at least for whatever reason the product no. is the same for both..?

Mario156090, 2022-10-28:
I will study about PCAP CAPTURE and attach later.

Mario156090, 2022-10-28:
[USB Razer.zip](https://github.com/openrazer/openrazer/files/9884758/USB.Razer.zip)
@z3ntu this capture works?

z3ntu, 2022-10-29:
@Mario156090 I don't see any Razer-style packets in there, but the packet capture is also 70MB so not sure what things you did while doing the capture that caused that much traffic. Especially for mice it's a good idea not to move the mouse around while doing the capture as that'll just add a lot of traffic to it.

And to be clear, you did this with Razer Synapse on Windows open and clicking around there, toggling different options?

Mario156090, 2022-10-29:
> @Mario156090 I don't see any Razer-style packets in there, but the packet capture is also 70MB so not sure what things you did while doing the capture that caused that much traffic. Especially for [...]

This capture is from Linux while I use the mouse. I should to capture from Windows and with synapse open?

z3ntu, 2022-10-30:
Yes, it's the data Razer Synapse software sends over USB that's interesting, nothing else.

Mario156090, 2022-10-31:
> 

Well, here a capture from Windows with Razer synapse open and I change some values from this.
[RazerPCAP.zip](https://github.com/openrazer/openrazer/files/9903702/RazerPCAP.zip)

z3ntu, 2022-10-31:
Yes, that's good now!

The device uses transaction id 0x1f, normal DPI and DPI stages feature, brightness with ZERO_LED, and `MATRIX_DIMS = [1, 1]` it looks like. Should be fairly simple to implement, if you want to take a shot at it.

z3ntu, 2022-10-31:
In general, replicate what other people did for similar device, e.g. https://github.com/openrazer/openrazer/commit/a573c91d8727296295e5a19233cf2c0668ccf712
Or better find a device that's even closer to your device, maybe `USB_DEVICE_ID_RAZER_DEATHADDER_V2_MINI`? And use common sense ;)

Blood-Turd, 2022-11-17:
> In general, replicate what other people did for similar device, e.g. [a573c91](https://github.com/openrazer/openrazer/commit/a573c91d8727296295e5a19233cf2c0668ccf712) Or better find a device that's even closer to your device, maybe `USB_DEVICE_ID_RAZER_DEATHADDER_V2_MINI`? And use common sense ;)

Hi. Mmm. I'm super new to both Linux and GitHub. I love coding but have zero practical knowledge about it, besides some very simple JavaScript old school ping pong game I coded following a step by step tutorial. So, I have no common sense in this case. I went to the link you shared and was lost in all that code, I don't know how to modify it to make Deathadder V2 Lite compatible with OpenRazer. 

It looks like it hasn't been implemented yet? Did Mario did it? I'm using Linux Mint 19 and it would be great to be able to remap my mouse keys through RazerGenie.

Thank you in advance.

Mario156090, 2022-11-19:
> Hi. Mmm. I'm super new to both Linux and GitHub. I love coding but have zero practical knowledge about it, besides some very simple JavaScript old school ping pong game I coded following a step by [...]

Me neither know how I can do that. I'm not developer or something.

PR #2055 (2023-04-29): Support for DeathAdder V2 Lite
Adds support for DeathAdder V2 Lite.

Based on information of DeathAdder V2 Mini, and data from #1912 and existing code for DeathAdder V2 Mini, but with transaction id 0x1f and idProduct  0x00a1. 

Tested locally with Polychromatic against lighting effects including brightness controls, DPI range, and standard polling rates.

It should resolve #1912
closed 2023-04-29 by z3ntu