openrazer/openrazer #1913: Support for Razer Kiyo Pro

tymokvo, 2022-09-22:
### Device Info

* Name: Razer Kiyo Pro
* Product Number: RZ19-03640100
* Model Number: RZ19-0364

### Device Information
<details>

```
Bus 002 Device 005: ID 1532:0e05 Razer USA, Ltd Razer Kiyo Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0e05 
  bcdDevice            8.21
  iManufacturer           1 Razer Inc
  iProduct                2 Razer Kiyo Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x05b4
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              0 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x0069
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0101 USB Streaming
[... 841 lines cut ...]
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay         0x0000
```
</details>

```
total 0
drwxr-xr-x. 2 root root 100 Sep 21 22:03 .
drwxr-xr-x. 4 root root 800 Sep 21 22:03 ..
lrwxrwxrwx. 1 root root  10 Sep 21 21:12 usb-Generic_Laptop_Camera_200901010001-event-if00 -> ../event11
lrwxrwxrwx. 1 root root   9 Sep 21 21:13 usb-Logi_Litra_Glow_2220FE007BM8-event-if00 -> ../event8
lrwxrwxrwx. 1 root root  10 Sep 21 22:03 usb-Razer_Inc_Razer_Kiyo_Pro-event-if00 -> ../event10
```

### Packet Captures

I can edit with this once I have my Windows VM working.

### Synapse Screenshots

Synapse on VM not working right now, I can edit with this.

lah7, 2022-09-22:
Just a heads up - this device was previously requested (#1665), but according to those packet captures, there wasn't anything Razer-specific found in the packet captures for OpenRazer to support. You're welcome to provide new captures in case this has changed _(new firmware revision, software update, etc)_

See also: https://github.com/openrazer/openrazer/issues/1910#issuecomment-1250229148

z3ntu, 2022-10-15:
Closing as duplicate. Unless you can prove me otherwise, there's nothing to support here from openrazer-side.

closed 2022-10-15 by z3ntu