openrazer/openrazer #1914: Support for Raptor 27 165hz

t56k, 2022-09-25:
### Device Info

* Name: Raptor 27 165hz
* Product Number: RZ39-03500100
* Model Number: RZ39-0350

### Device Information
```
Bus 003 Device 009: ID 1532:0f28 Razer USA, Ltd Razer Raptor 27 165Hz
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f28
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Raptor 27 165Hz
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1

```

```
lrwxrwxrwx 1 root root   9 Sep 25 07:42 usb-AONE_Varmilo_Keyboard-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Sep 25 07:42 usb-AONE_Varmilo_Keyboard-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root  10 Sep 25 08:58 usb-Generic_Raptor_27_165Hz_USB_Audio-event-if03 -> ../event18
lrwxrwxrwx 1 root root   9 Sep 25 07:42 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Sep 25 07:42 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Sep 25 09:29 usb-Razer_Razer_Raptor_27_165Hz-event-if01 -> ../event17
```

oleg-the-developer, 2024-11-18:
@z3ntu hey could you please let me know what pcap data is needed to add this monitor ?

z3ntu, 2024-12-07:
@oleg-the-developer You can take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself.

z3ntu, 2024-12-07:
Or just for pcap, with that I could give you some hints what you need to add in the driver.

For that we need pcaps of the USB traffic of Razer Synapse on Windows, see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol how to capture.

zoey-marie, 2025-06-19:
Hey. I don't really know what I'm doing but I'd like support for this as well. I cloned the github repo and basically just added in additional entries for the Raptor 27 for the 165hz version. I'm pretty sure the lighting is the same, just the monitor refresh rate is different. But I don't know how to compile the cloned repo to add in the changes with the installed program? Maybe that's putting the cart before the horse but I can't imagine there's more to this than just adding an entry for the device ID. Again, though, I don't really know what I'm doing. Any help would be appreciated.

z3ntu, 2025-07-06:
@SmallestMeow We also have this wiki page, this tells you how to go from source to compile packaged so you can test it: https://github.com/openrazer/openrazer/wiki/Building-a-package

SmallestMeow, 2025-07-22:
> Not sure what I can suggest without a more detailed description.
> 
> It should be written in some of the docs, first make sure that the driver has actually bound, the udev rules run a script to bind the - in your case - razeraccessory driver to the USB device. Then sysfs files should show up.
> 
> Once they're there, the daemon can pick up the device and then it should show up in the daemon output, and be usable from e.g. polychromatic

Hey... so idk if you all added support for this monitor, I assume not since the issue is still open and it's not on the Readme, but if you didn't then the edits are working on my end. It shows up and the lighting can be controlled in Polychromatic. Did you all fix it on your end? Or should I do something to support other people getting the updated functionality? Thanks!

z3ntu, 2025-07-29:
@SmallestMeow If you have changes to OpenRazer that make this device work, then please send a pull request with them! This device (1532:0f28) is certainly not supported yet in OpenRazer master.

SmallestMeow, 2025-07-31:
@z3ntu  Sounds good!! I'm pretty inexperienced with GitHub stuff, so it might take me a minute to figure out how to do it right, but yeah it's all working on my end. I just added a couple entries in a few files copying the non-165hz one and just adding the device ID and using a different label; nothing fancy. I just looked and it looks like the master has changed a bunch since I made my edits, so I need to figure out how to merge it all on my end and probably add in the edits in a few more places. I won't have time for a few days, but I'll try and figure it out when I can! Thanks!

closed 2026-11-16 by z3ntu