openrazer/openrazer #1917: DKMS Install Failed to read file's magic number

alexonpeace, 2022-09-29:
`$ sudo dkms install -m openrazer-driver/3.4.0
Sign command: /usr/lib/modules/5.19.12-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
readelf: Error: /usr/lib/modules/5.19.12-arch1-1/build/vmlinux: Failed to read file's magic number
readelf: Error: /usr/lib/modules/5.19.12-arch1-1/build/vmlinux: Failed to read file's magic number

Building module:
Cleaning build area...(bad exit status: 2)
KERNELDIR=/lib/modules/5.19.12-arch1-1/build make driver...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.12-arch1-1 (x86_64)
Consult /var/lib/dkms/openrazer-driver/3.4.0/build/make.log for more information.`

alexonpeace, 2022-09-29:
i don't have secure boot on tho
it's off, always been off

Karagra, 2022-09-29:
I'm prob wrong but doesn't this happen when you are missing the kernel headers?

z3ntu, 2022-09-30:
Check the file `/var/lib/dkms/openrazer-driver/3.4.0/build/make.log` then

closed 2023-03-16 by github-actions[bot]