openrazer/openrazer #1923: Razer DeathStalker V2

simplyroman, 2022-10-10:
### Device Info

* Name: DeathStalker V2
* Product Number: RZ03-04500400-R3G1
* Model Number: RZ03-0450

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

`Bus 003 Device 004: ID 1532:0295 Razer USA, Ltd Razer DeathStalker V2
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0295 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer DeathStalker V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1`

```

```
Please post the output of
    ls -lah /dev/input/by-id/

`lrwxrwxrwx. 1 root root   9 10. Okt 11:22 usb-Razer_Razer_DeathStalker_V2-event-if01 -> ../event5
lrwxrwxrwx. 1 root root   9 10. Okt 11:22 usb-Razer_Razer_DeathStalker_V2-event-kbd -> ../event3
lrwxrwxrwx. 1 root root   9 10. Okt 11:22 usb-Razer_Razer_DeathStalker_V2-if01-event-kbd -> ../event4
lrwxrwxrwx. 1 root root   9 10. Okt 11:22 usb-Razer_Razer_DeathStalker_V2-if02-event-mouse -> ../event6
lrwxrwxrwx. 1 root root   9 10. Okt 11:22 usb-Razer_Razer_DeathStalker_V2-if02-mouse -> ../mouse0`

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

simplyroman, 2022-11-06:
pcap added

z3ntu, 2022-11-09:
@simplyroman Not much in the pcap except two brightness changes, but should be ok regardless

simplyroman, 2022-11-09:
@z3ntu Thanks. Appreciate your efforts.

z3ntu, 2022-11-09:
Ah, forgot to add. pcaps show Synapse is using transaction ID 0x3f, and uses standard brightness command with ZERO_LED.

But there were also multiple devices connected during the pcap so not 100% sure it's from the correct one.

AZeroEight, 2022-11-16:
I tried to run a Wireshark on my windows machine and did some capture if it helps. It appears to me the ID is actually 0x1F. (if it is the second byte of the data fragment).

simplyroman, 2022-11-16:
@AZeroEight I think that would help. My knowledge with wireshark is really low.

AZeroEight, 2022-11-16:
Here you go. The device in question is 2.8.0. I tried changing colors, lighting effects and brightness. I also tried to edit the driver, compile it and run it on my system but so far the driver does not accept the device: "Response doesn't match request". Would be great to know what the bare minimum is to start off.

[deathstalker_v2_2.zip](https://github.com/openrazer/openrazer/files/10021831/deathstalker_v2_2.zip)

AZeroEight, 2022-11-16:
**I actually got most of it working already** - I need to verify the available settings in Synapse later. The "Game Mode" button does not work yet and I haven't tried Macro recording.

Once I am finished - how can I share my changes with you? Oh and how do I actually get the correct picture? I've used the one from another device so far.

[image]

AZeroEight, 2022-11-17:
I've created a pull request for my changes to be applied and get the keyboard fully working.

simplyroman, 2022-11-17:
@AZeroEight Thanks a lot. Can't wait till it is beeing officially released.

PR #1953 (2022-11-20): Add support for Razer DeathStalker V2
I've implemented the necessary changes for the Razer DeathStalker V2. Everything is working except macro recording. I've never used that function and from what I've read you need the macro keys 1-5 to get it working in openrazer but they do not exist on that keyboard.

The picture used is wrong as I don't know the URL of the correct one - could anybody tell me how I can look that information up?

Resolves #1923
closed 2022-11-20 by z3ntu