openrazer/openrazer #1925: Support for Ornata v3

ddB0515, 2022-10-11:
### Device Info

* Name: Razer Ornata v3
* Product Number: RZ03-04460300-R3W1
* Model Number: RZ03-0446

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 029: ID 1532:028f Razer USA, Ltd Razer Ornata V3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028f 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root  10 Oct 11 16:39 usb-Razer_Razer_Ornata_V3-event-if01 -> ../event20
lrwxrwxrwx 1 root root   9 Oct 11 16:39 usb-Razer_Razer_Ornata_V3-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Oct 11 16:39 usb-Razer_Razer_Ornata_V3-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root  10 Oct 11 16:39 usb-Razer_Razer_Ornata_V3-if02-event-mouse -> ../event21
lrwxrwxrwx 1 root root   9 Oct 11 16:39 usb-Razer_Razer_Ornata_V3-if02-mouse -> ../mouse0

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

ddB0515, 2022-10-17:
Attaching capture from Wireshark
[Razer-Ornata-v3-USBCapture.pcapng.zip](https://github.com/openrazer/openrazer/files/9799834/Razer-Ornata-v3-USBCapture.pcapng.zip)

ddB0515, 2022-10-18:
@z3ntu have you checked if pcap is valid? If not to make new one
Let me know. Thanks

z3ntu, 2022-10-29:
Yes, looks good!

The device uses transaction id 0x1f, `MATRIX_DIMS = [1, 9]`, the rest is probably also very standard, but the packet capture doesn't include e.g. brightness change. Not necessary though as those are pretty surely like all other devices.

ddB0515, 2022-10-29:
Created PR with support for Ornata v3
Brightness working and only Starlights were removed as is not supported everything else works 
Thanks for help :)

Tankeran, 2023-04-24:
I have the same keyboard using on steam deck. Since i have no experience could you add native support on the new openrazer version?

commit fe87b2d5 (2024-09-24): Add support for Razer Ornata V3 [028F]

Fixes #1925
PR #2233 (2024-09-24): Add support for Razer Ornata V3 [028F]
This is to add support for the `028f` PID variant of the Razer Ornata V3, following the work from #2182.

#2182 originally contained support for both `02a1` and `028f` PID variants. While the former variant was reported to work without issues by different users, one user, @ddB0515, [reported issues and errors](https://github.com/openrazer/openrazer/pull/2182#issuecomment-1933990284) with the latter despite being able to set chroma. Support for the `02a1` variant was merged instead, but these issues with `028f` have yet to be narrowed down and no other use [... cut]
closed 2024-09-24 by z3ntu