openrazer/openrazer #1926: Support for Basilisk V3 Pro

0xADF, 2022-10-13:
### Device Info

* Name: Razer Basilisk V3 Pro
* Product Number: RZ01-04620100-R3A1
* Model Number: RZ01-0462

### Device Information
```
lsusb -d 1532: -v

Bus 003 Device 006: ID 1532:00ab Razer USA, Ltd Razer Basilisk V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00ab 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk V3 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-04d9_USB_Keyboard-event-if01 -> ../event19
lrwxrwxrwx 1 root root   9 Oct 13 14:22 usb-04d9_USB_Keyboard-event-kbd -> ../event3
lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-event-if01 -> ../event23
lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-event-mouse -> ../event20
lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-if01-event-mouse -> ../event22
lrwxrwxrwx 1 root root   9 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-if02-event-kbd -> ../event24
lrwxrwxrwx 1 root root   9 Oct 13 14:22 usb-Razer_Razer_Basilisk_V3_Pro_000000000000-mouse -> ../mouse0
```

### Packet Captures
Can provide if necessary

### Synapse Screenshots
[image]
[image]
[image]
[image]
[image]

TherionRO, 2022-10-13:
+1 , I also need support for the V3 Pro . Thanks!

DavidescuEMI, 2022-10-13:
+2 really really need support for this model with openrazer

Mathewthemvp, 2022-10-13:
+3 Lmao also here

z3ntu, 2022-10-29:
@0xADF Thanks!

For anyone wanting to take a shot at implementing support for this device, use transaction id 0x1f, the rest looks pretty standard for DPI, DPI stages, poll rate, brightness is via ZERO_LED. And probably `MATRIX_DIMS = [1, 12]` from what I see

0xADF, 2022-12-01:
Seems to be working ok for Basilisk V3 Pro. Many thanks!

dekomote, 2022-12-18:
I tested this one out, it seems to be working properly. Battery readings are consistent with synapse, lighting works per spec, changing all parameters including scroll smart reel, everything is pretty much the same as synapse. Tested with original 1.00 firmware as well as the newer released 1.01 firmware.

arrowj, 2022-12-18:
@dekomote when you said it seems to be working properly, does that mean in an upcoming release? My brand new Basilisk V3 Pro (ID 1532:00AB) is not recognized. The device can of course connect via Bluetooth, dongle, and wired. The profile of button assignments I set up in Windows work with or without OpenRazer installed, but on Fedora 37 with the latest OpenRazer and Polychromtic no joy on the LED control or battery life. Latest firmware 1.01.

dekomote, 2022-12-18:
> @dekomote when you said it seems to be working properly, does that mean in an upcoming release? My brand new Basilisk V3 Pro (ID 1532:00AB) is not recognized. The device can of course connect via [...]

Sorry I wasn't clear, I was referring to the pull request linked above #1964

PR #1964 (2023-01-15): Add support for Razer DeathStalker V2 Pro Wired/Wireless and for Basilisk V3 Pro Wired/Wireless
I've implemented the changes for the DeathStalker V2 Pro Wired/Wireless and for the Basilisk V3 Pro Wired/Wireless.
Everything seems to be ok... except macro recording for DeathStalker V2 Pro.

Resolves #1907
Resolves #1926
closed 2023-01-15 by z3ntu