openrazer/openrazer #1927: No automatic startup

marcoszy, 2022-10-13:
### Distribution

Kubuntu 22.04 / KDE 5.25.5

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

Hi there,

I can´t make Polychromatic start automatically with OS despite the "start the tray applet when I log on" setting is checked.

I also tried to put a .desktop file inside ~/.config/autostart, but the program is not started when the system starts.

Any idea about how could I solve this, please?

Thanks in advance.

lah7, 2022-10-13:
Try checking the System Monitor in case it's running in the background but not visible.  If a `polychromatic-tray-applet` process is running, you can specify a few seconds delay in `~/.config/polychromatic/preferences.json`. I use KDE and set it to `4` because if it loads too early, it renders weird.

If the tray applet simply doesn't start at all, run from the terminal and see if there's an error message:

    polychromatic-tray-applet

Also, this is the wrong repo. [polychromatic/polychromatic](https://github.com/polychromatic/polychromatic) would be the correct place, unless it is a problem caused by OpenRazer.

marcoszy, 2022-10-13:
Hello,

Thank you very much and sorry for posting in the wrong place.

So, I tried everything, without success, unfortunately.

But, running from the terminal produced the below output:

--
/usr/bin/polychromatic-tray-applet:224: DeprecationWarning: Gtk.ImageMenuItem.set_image is deprecated
  item.set_image(img)            # GTK3: Deprecated, no replacement!
--

I have no idea about this, however.

lah7, 2022-10-13:
No problem. The good news is that it is running. Do you have a "system tray" widget added to your panel? It could be hidden in the "overflow" menu.

You can try changing the mode used to render, and restart the tray applet. If you have the Controller installed, you can do that via Preferences. By file, it's in `~/.config/polychromatic/preferences.json` with the `tray` → `mode` key. 

- 0 = Default _(I think it's currently Ayatana)_
- 1 = AppIndicator
- 2 = GTK Status (Legacy)
- 3 = Ayatana Indicators

marcoszy, 2022-10-14:
Thanks for your kindness, and sorry for bothering you. :)

Well, I really don't know what's happening here. :(

There's no icon hidden in the system tray. And, I have already changed the mode in the JSON file. Nothing works here, unfortunately.

The only way to start it is manual.

lah7, 2022-10-15:
I tried Kubuntu 22.04.1 (💿 live session) and couldn't reproduce. The tray applet appears a few seconds after login (without OpenRazer installed)

If you say it works by manually starting, the custom entry in `~/.config/autostart` entry might want deleting. By default, the app starts a process called `polychromatic-helper --autostart` (in `/etc/xdg/autostart`) and that launches the tray applet if enabled in `preferences.json`. `polychromatic-tray-applet` will take over another instance if one was already running.

If it's the case that it can't launch because `openrazer-daemon` isn't ready on your system for some reason, you may want to create a custom shell script and put the commands in there:
```sh
openrazer-daemon
sleep 3
polychromatic-tray-applet
```

I'll close the issue here since it's not about OpenRazer. If you find a bug or still need help with this, feel free to raise the issue on [polychromatic's repo](https://github.com/polychromatic/polychromatic).

closed 2022-10-15 by lah7