openrazer/openrazer #1930: Support for Nari Ultimate

daedalusSecurity, 2022-10-18:
### Device Info

* Name: Nari Ultimate
* Product Number: 051a
* Model Number: ?

### Device Information
```
Bus 001 Device 015: ID 1532:051a Razer USA, Ltd Nari Ultimate
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051a Nari Ultimate
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari Ultimate
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
[... 359 lines cut ...]
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/

usb-Razer_Razer_Nari_Ultimate-event-if05 -> ../event23

```

nightsky30, 2022-10-18:
Please see #974

z3ntu, 2022-10-29:
Duplicate of #974

closed 2022-10-29 by z3ntu