openrazer/openrazer #1932: Screensaver code mishandles mouse brightness when the device is reconnected

dkonigsberg, 2022-10-19:
First of all, I'm not exactly sure how to title this bug, so feel free to change the title as appropriate.

For software, I'm using openrazer-daemon 3.4.0 on Linux (Pop!_OS 22.04) with a Razer DeathAdder V2 mouse.

This Linux machine is connected to its peripherals through a KVM, to share them with another computer.  What I've stumbled across is a bug in the way the screensaver light brightness handling code behaves in combination with the mouse disconnect/reconnect events that happen because of the KVM.  Basically what happens, is when I switch back to the Linux machine after being off it for a prolonged period, it cranks the light brightness on my mouse all the way up to 100%.  (I normally keep it set at 15%.)

Here's the steps to reproduce:

1. Switch away from the Linux machine via the KVM, thus disconnecting the mouse
2. Wait long enough for the screensaver to kick in on the now-hidden Linux machine
3. Switch back to the Linux machine via the KVM, thus reconnecting the mouse
4. Observe the following sequence of events:
    1. Mouse reattaches and initializes at correct brightness
    2. Screensaver detects relevant events and shuts off
    3. Mouse brightness immediately goes all the way up to 100%

Based on my investigation of the code and logs, here's what I think is happening:

- When the screensaver activates, openrazer-daemon attempts to save the current mouse brightness state and then shut off the mouse lights. However, the mouse has already been disconnected at this point, so this does not work correctly.
- When the screensaver deactivates, openrazer-daemon attempts to restore the mouse brightness. However, because it failed to save it correctly, it just gets default values (which are 100%) and applies those instead.

In my particular case, the relevant code is here:
https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L2537-L2564
However, its not clear whether it makes more sense to fix that code, or to fix the code that calls it.

Regardless, by kludging the default values for brightness in that code on my system, I've been able to implement a localized workaround.

(This started on an issue thread for a related project: https://github.com/polychromatic/polychromatic/issues/186#issuecomment-1283004911)

z3ntu, 2022-11-05:
Your explanation sounds reasonable. What we could do is instead of directly saving (and restoring) the brightness in suspend/resume callbacks is use the one saved by persistent storage feature.

Do you want to take a shot at fixing this? If you have any questions I'm here to help of course.

dkonigsberg, 2022-11-11:
Unfortunately, I'm going to have to take a pass on working this issue myself.  I'm not really a Python programmer, so I'm not comfortable jumping in and doing much more than basic kludges.

z3ntu, 2022-11-20:
Could you please test #1957? In my testing this seems to fix this but would be good that you confirm as well.

dkonigsberg, 2022-11-21:
> Could you please test #1957? In my testing this seems to fix this but would be good that you confirm as well.

Just tested locally, and I can confirm that this fixes the issue.

PR #1957 (2022-12-06): Refactor suspend/resume handling
Since we now have persistence feature we can get rid of most of the custom suspend/resume functions implemented for every single device and use the persistence feature to do the same instead.

Fixes #1932 (and maybe some other issues?)