openrazer/openrazer #1933: My devices are unrecognized by openrazer

MartinYang1, 2022-10-20:
### Distribution

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

Hi, I have a razer viper mouse and a razer huntsman tenkeyless TE keyboard and they are unrecognized by openrazer even though on the website, it says they are supported. The device numbers are 1532:026B and 1532:0078.

z3ntu, 2022-10-21:
0078 should be supported, the other device is tracked in #1681

Can you provide output of `sudo modprobe razerkbd` and `systemctl --user status openrazer-daemon.service` ?

MartinYang1, 2022-10-21:
yes, now 0078, my razer viper, is recognized. however the keyboard is unrecognized. Could it be because my openrazer version is 3.2.0? My openrazer-meta is already the newest version, but the openrazer-src newest version is 3.4.0
[image]

MartinYang1, 2022-10-22:
i looked at openrazer's website and readme.md and they don't seem to support razer huntsman v2 tenkeyless. they support huntsman te or huntsman v2 analog, but not this

z3ntu, 2022-10-27:
As mentioned, the 026B device is not yet supported. You can subscribe to issue #1681 to get notified when it will be.

As the other device is detected fine (so no setup issue or anything), I'll close this issue.

closed 2022-10-27 by z3ntu