openrazer/openrazer #1937: Polychromatic does not detect my devices

marcoszy, 2022-10-26:
### Distribution

Kubuntu 22.04

### Support

- [ x ] Have you tried rebooting?
- [ X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hey there!

I was trying to figure out what is happening here, and I've opened an issue here: https://github.com/openrazer/openrazer/issues/1927#issuecomment-1279715910

But, today the problem became worse. :(

Trying to fix the issue I've uninstalled both Polychromatic and OpenRazer. After that I installed both.

But now Polychromatic is not detecting my Razer devices (unrecognized device). Both the keyboard and mouse are listed with the lsusb command, moreover.

The command `sudo dkms install openrazer-driver/3.4.0` returns the below error:

"headers for kernel 5.17.0-1020-oem cannot be found."

But I really don't know if this is the problem and/or if I am missing something.

Any help would be greatly appreciated! :)

z3ntu, 2022-10-27:
I guess you're missing the package `linux-headers-5.17.0-1020-oem` , not sure if there's some meta package you can install that makes this installable (and for future versions also).

marcoszy, 2022-10-27:
Hum, I will check that. Thanks.

kyle-belle, 2022-11-04:
My device also is no longer recognized (razer blade 15 2019) running Zorin OS 16.2 [Linux 5.18.15-051815-generic x86_64]

closed 2023-04-10 by github-actions[bot]